zephyrproject-rtos/zephyr #8784: Introduce cache infrastructure

pizi-nordic, 2018-07-06:
TODO:

- [ ] Create system-level API for cache operations:
  - [ ] I-Cache: enable/disable, invalidate.
  - [ ] D-Cache: enable/disable, writeback, invalidate, writeback + invalidate.
 
- [ ] Create system-level memory allocation primitives for DMA memory.
  - [ ] Create macro ensuring cache-line alignment of static data structures.

- [ ] Enable memory caching in MPU:
  - [ ] Flash: Write-Through
  - [ ] RAM: WriteBack + Write Allocate.

- [ ] Update platform code and drivers to use Cache API.

Optional:
  - [ ] Create API for dynamic memory allocation (aligned to cache line, size always multiple of cache line).
  - [ ] Create API for marking memory block non-cacheable.

ioannisg, 2019-05-14:
> Enable memory caching in MPU:
 Flash: Write-Through
 RAM: WriteBack + Write Allocate.

@pizi-nordic @aurel32 could we say this is done for ARM?

aurel32, 2019-05-16:
> > Enable memory caching in MPU:
> > Flash: Write-Through
> > RAM: WriteBack + Write Allocate.
> 
> @pizi-nordic @aurel32 could we say this is done for ARM?

@ioannisg Yes, I confirm this is done for ARM.

>  Create API for marking memory block non-cacheable.

This is also done for static allocations with `__nocache`. It's not clear if it should also be done for dynamic allocations.

andrewboie, 2020-08-13:
related: https://github.com/zephyrproject-rtos/zephyr/issues/2647

nashif, 2024-02-08:
closing as done

closed 2024-02-08 by nashif