zephyrproject-rtos/zephyr #8791: Request usb_dc_stm32 to support OTG_HS port on STM32F4 SoCs

jli157, 2018-07-06:
The OTG_HS port on STM32F4 SoCs can support both full-speed USB transmission with its integrated transceiver at 12Mbps, and high-speed USB transmission at 480Mbps with the external USB phy.  So, the pins PB14 and PB15 from OTG_HS port can be directly used as a full-speed USB interface, providing same functionality as the OTG_FS port from PINs PA11 and PA12. To support OTG_HS as a full-speed USB port, the device driver usb_dc_stm32 needs to be changed to use the corresponding registers for OTG_HS port.

qianfan-Zhao, 2018-07-07:
but seems there doesn't has a nucleo or discovery board has a built-in USB HS phy.

jli157, 2018-07-10:
STM32F723 has a built-in HS port. So, this request can be verified with STM32F7 boards.