zephyrproject-rtos/zephyr #8804: esp32 cannot get character from UART port

greatmay, 2018-07-09:
I have installed  Zephyr OS  in  ESP32 wroom 32 device .
It seem that uart receiving it is not work after I enter character in serial terminal then CPU cannot jump execute instruction set
message error as shown below
** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x3ffb0a58
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740
Current thread ID = 0x00000740
Faulting instruction address = 0xdeaddead
 ** FATAL EXCEPTION
 ** CPU 8 EXCCAUSE 28 PS 0x00060120 PC 0x80080eae VADDR 0x00000748
 **  A0 0x70252053  SP 0x3ffb07f3  A2 0x56207025  A3 0x52444441
 ** LBEG 0x45206425 LEND 0x20555043 LCOUNT 0x202a2a20
 ** SAR 0x41434358
Current thread ID = 0x202a2a20
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
 ** Spurious INTERRUPT(s) 0x00018060, INTENABLE = 0x00000740

nashif, 2018-08-27:
afaik, this is not supported on esp32 yet.

nashif, 2019-01-30:
duplicate of #3981

closed 2019-01-30 by nashif