zephyrproject-rtos/zephyr #8810: Cannot flash board stm32f4_disco

finikorg, 2018-07-09:
Flashing board http://docs.zephyrproject.org/boards/arm/stm32f4_disco/doc/stm32f4_disco.html

Maybe I have a slightly different revision: 

> STM32F407G-DISC1

```
[100%] Flashing stm32f4_disco
Using runner: openocd
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Open On-Chip Debugger 0.10.0-g7852ae77-dirty (2018-05-11-23:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Error: open failed
in procedure 'init' 
in procedure 'ocd_bouncer'

error: command exited with status 1: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f <path>arm/stm32f4_disco/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase <path>samples/hello_world/outdir/stm32f4_disco/zephyr/zephyr.bin 0x8000000' -c 'reset halt' -c 'verify_image <path>samples/hello_world/outdir/stm32f4_disco/zephyr/zephyr.bin 0x8000000' -c 'reset run' -c shutdown
run as "west -v ... flash ..." for a stack trace
[100%] Built target flash
```

erwango, 2018-07-30:
@finikorg , this same issue was identified in https://github.com/zephyrproject-rtos/zephyr/issues/7144.
Problem is we're not able to handle multiple board revision.
So either we need to fix ST-Link version and older revision won't work, either we add it as a new board, but this is large code duplication for minimal gain.

One option is to document this issue in board .rst file.

finikorg, 2018-08-09:
@erwango Maybe we can add a new board but try to source as much as possible from the "general" section?

erwango, 2018-08-09:
@finikorg , not sure what you mean by "general" section. 
My main issue is to duplicate minimum possible code and information, if you manage to do this, we can have a look.

tadmorgan, 2018-09-29:
I'm trying to evaluate several different RTOS's on the stm32f407g-disc1 and am running into this issue when trying out Zephyr on v1.12.0. Has there been in resolution? Is there a recommended work around that provides full support? Not supporting the currently available boards seems to be a poor choice from POV of someone considering the OS...

erwango, 2018-12-13:
Could be solved using `board_runner_args` (eg: board_runner_args(nrfjprog "--nrf-family=NRF51") in board.cmake

erwango, 2018-12-19:
@finikorg , can you have a try with upstream openocd?
Seems that they(ve done some changes on interface/stlink-v2.cfg/ interface/stlink-v2-1.cfg which are both redirecting to interface/stlink.cfg.
I hope this should solve our issue

nashif, 2019-01-30:
@finikorg @erwango any update on this? is this an openocd issue or some configuration we have in the tree?

erwango, 2019-01-30:
@nashif, there are actually 2 revisions of the board, each one was working with a different openocd config. According to what I can see in recent openocd, they should now both work with existing zephyr configuration. I hope @finikorg can test and confirm.

erwango, 2019-01-31:
@finikorg, can you test Blinky sample? I wonder if ST Link VCP is wired on this board.

finikorg, 2019-01-31:
@erwango `blinky` works, also tested make debug working, maybe magic DTS serial port name configuration is worng?

erwango, 2019-01-31:
@finikorg, actually, unlike other disco or nucleo boards, you need to add some wires first to get ST Link VCP serial Rx Tx matching STM32 serial port. The SW modification would be to remove console as it is not available by default on boards

erwango, 2019-01-31:
Anyway, can we close the issue, now that you confirm you are able to flash the board?

finikorg, 2019-01-31:
> @finikorg, actually, unlike other disco or nucleo boards, you need to add some wires first to get ST Link VCP serial Rx Tx matching STM32 serial port. The SW modification would be to remove console as it is not available by default on boards

This makes the board almost unusable, should we remove the board from the supported board list?

finikorg, 2019-01-31:
Otherwise the issues might be closed, should I close it?

erwango, 2019-02-01:
> This makes the board almost unusable, should we remove the board from the supported board list?

This is actually the same status as some other STM32 based, non ST boards, such as stm32_min_dev, olimexino_stm32 and alikes, on which you need to connect a serial cable on some pins to use  Not sure this is a sufficient reason to remove them.
Anyway, I'm pushing a PR to clarity board documentation

> Otherwise the issues might be closed, should I close it?
Done

closed 2019-02-01 by erwango