zephyrproject-rtos/zephyr #8820: wifi_winc1500 driver socket id stored in net_context->user_data may be overwritten at socket layer

GAnthony, 2018-07-10:
While looking through net_context.h for a place to store the net offload driver's (internal) socket handle, considering the net_context->user_data field, and seeing what was done in wifi_winc1500.c, I noticed that socket.c reuses the same user_data for flags, potentially overwriting the driver's socket handle for that net_context.

From drivers/wifi/winc1500/wifi_winc15000.c:
static int winc1500_get(sa_family_t family,
			enum net_sock_type type,
			enum net_ip_protocol ip_proto,
			struct net_context **context)
{
	struct socket_data *sd;
[...]
	(*context)->user_data = (void *)(sint32)socket(family, type, 0);

Later, if socket API is used over net_context (with this offload driver), then:
subsys/lib/sockets/socket.c:
static inline void sock_set_flag(struct net_context *ctx, u32_t mask,
				 u32_t flag)
{
	u32_t val = POINTER_TO_INT(ctx->user_data);
	val = (val & ~mask) | flag;
	(ctx)->user_data = INT_TO_POINTER(val);
}

Can sockets use the available bits of the net_context->flags parameter instead, or would it be better to have an optional, more clearly named, void *offload_context for NET_OFFLOAD drivers?

jukkar, 2018-07-10:
So far there has not been a conflict for user_data variable between various parts of the system. The user_data is for caller specified data if net_context is used directly. For BSD sockets it is the only one using net_context so it has been using it as is. Perhaps it would be best if you introduce offload_context for offload use.

GAnthony, 2018-07-10:
OK, I can add an offload_context field, guarded by CONFIG_NET_OFFLOAD, and also update the winc1500 driver, and propose a PR.

pfalcon, 2018-07-12:
Per https://github.com/zephyrproject-rtos/zephyr/blob/master/include/net/net_context.h#L800, etc. user_data is outer (client) facing API parameter, in other words, it fully belongs to client application and not the system. So, using that field in the driver is clearly a non-intended usage.

Sockets API uses that param/field because it exactly a client of native API, and it doesn't expose such a param in its API, so it's up to it to use it for its own needs.

> Can sockets use the available bits of the net_context->flags parameter instead

Socket layer initially conceived to be as much as possible independent from the underlying implementation, for example, not having a knowledge which bits in net_context->flags are "available" (what if Sockets API assigned some sense to them, but later new flags are added on the native stack level?). That's why user_data was used instead (as per above, it was possible to be 100% sure that user_data is up to sockets layer to use). Now that Sockets API is first-class citizen, we probably can move its flags down to net_context level. But in either case, the drivers or any other part of the IP stack which can be accessed via native API should not use user_data for its own needs (it belongs to API user). So, wifi_winc1500 issue should be fixed anyway, and the simplest way is not touch user_data usage by sockets API so far.

GAnthony, 2018-07-12:
> So, wifi_winc1500 issue should be fixed anyway, and the simplest way is not touch user_data usage by sockets API so far.

Yes, so this issue should be fixed by PR #8835.

PR #8835 (2018-07-18): drivers: wifi: winc1500: Use offload_context instead of user_data.
The wifi_winc1500 driver's socket id is stored in
net_context->user_data, which may be overwritten later at
the socket layer, which also uses the net_context->user_data
field to store socket flags.

This patch introduces a dedicated offload_context field
for use by offload drivers, and updates the wifi_winc1500 offload
driver to use this field instead of user_data.

Fixes #8820
commit 42f51ef0 (2018-07-18): drivers: wifi: winc1500: Use offload_context instead of user_data.

The wifi_winc1500 driver's socket id is stored in
net_context->user_data, which may be overwritten later at
the socket layer, which also uses the net_context->user_data
field to store socket flags.

This patch introduces a dedicated offload_context field
for use by offload drivers, and updates the wifi_winc1500 offload
driver to use this field instead of user_data.

Fixes #8820
closed 2018-07-18 by nashif