zephyrproject-rtos/zephyr #8827: CMake overhaul and architecture

tejlmand, 2018-07-10:
This is an umbrella issue for a series of changes to the current CMake usage.

The purpose is to have a holistic view on the complete CMake build system in order to link multiple tasks together, where each change might seems a minor issue, but where the complete Zephyr CMake build architecture could be improved.

In general, it is better to have few, well organized / structured libraries, and populate those in a well defined way.
Those libraries should follow the general KConfig and folder structure, so that e.g. a subsystem folder corresponds exactly to a single library, to which dependencies can be clearly defined.

Example: Subsystem Net creates a library target: subsys_net (file name: libsubsys_net.a)

It must also be remembered that there is a difference between working in vi(m)/emacs/etc. and IDE's such as Eclipse.
Certain elements, such as target / libraries are more visible in an IDE.

Furthermore it suggest to avoid the wrapping of CMake functions, and instead use plain CMake whenever possible. Currently, many wrapped cmake functions limits the feature set of CMake and thus spawns additional commands (=> increased maintenance + additional API to learn for new comers).
Instead, functions that can assist the user, before using the plain CMake function should be preferred.

- [ ] Reduce number of libraries / library content cleanup to improve  developer experience in IDE
  * Issue: #8826 : Number of libraries presented
  * Issue: #8825 : File placement in libraries
  * PR: #8451 

- [ ] Bug related to too many libraries when enabling application memory space:
  * Issue: #7703
  * PR:

- [ ] Whole-archive / start-group / end-group link command
  * Issue: #8441 
  * Issue: #6961 
  * PR: #8451 

- [ ] CMake wrapped functions
  * Issue: #8439 
  * PR: #8442 

- [ ] General clean-up of linker flags
  * Issue: #8440
  * PR: #8445

SebastianBoe, 2018-08-20:
Hi,

"Bug related to too many libraries when enabling application memory space:"

should be removed as the bug has now been resolved.

SebastianBoe, 2018-08-20:
Also,

start-group and end-group has been removed, so

"Whole-archive / start-group / end-group link command"

can be changed to

"Whole-archive command"

tejlmand, 2018-08-21:
> Whole-archive / start-group / end-group link command

Partly true.
When --whole-archive is used, then --start-group / --end-group is not needed.
--whole-archive includes the whole archive, and thus any circular dependencies of course disappears.
But when removing the --whole-archive, then the problem which --start-group / --end-group is actually addresses re-appears.

Removing --whole-archive doesn't by itself ensure correct link libraries dependencies, and unless correct link dependencies (including circular dependencies) is introduced in CMake, then linking will fail, as the linker will search for down the chain.
Example:
libA depends on libB
libB also depends on libA
this dependency can be handled by:
--start-group libA libB --end-group
or when correctly described in CMake you'll get:
libA libB libA
during linking, in which case all symbols can be resolved.

SebastianBoe, 2018-08-21:
Sorry, what I was trying to say, is that we can update the issue description. It is stating that we need to get rid of "Whole-archive / start-group / end-group". But "start-group / end-group" has been gotten rid of, so now the issue description should just say that we need to get rid of "--whole-archive".

I am well aware of what flags do.

tejlmand, 2018-08-28:
ok.

Updated, and made own check-mark, as I believe we don't want the --start-group / --end-group reintroduced when removing the --whole-archive.

clamattia, 2024-12-06:
I am a bit dissatisfied with the cmake situation.
For example:
```
function(zephyr_library_include_directories)
  target_include_directories(${ZEPHYR_CURRENT_LIBRARY} PRIVATE ${ARGN})
endfunction()
```

How should someone know, that it is `PRIVATE`?
Why do we only have a wrapper for `PRIVATE` and not for example `INTERFACE`?
Why not use plain cmake (inlining the above function), which would increase readability for people familiar with cmake?

similar with the `zephyr_library` and `zephyr_library_named` etc etc...

What is the current situation on this subject @tejlmand ?

yashi, 2024-12-06:
@clamattia #8439