zephyrproject-rtos/zephyr #8833: OpenThread: Minimal Thread Device (MTD) option is not building

ramonhpr, 2018-07-10:
Hi guys,

I'm working in a project where I have to use the minimal thread device (MTD) from openthread and I am using the zephyr as a full thread device (FTD) and is working fine with 2 boards nrf52840 but when I set the MTD option the bellow problem appears.
I am working in a fork a solution for this problem. With my changes in CmakeList the MTD was compiled properly. But the sample app `echo_client` was not sending through udp to my server (which was working normally with FTD). So my solution was to bump to the newest version of openthread.

``` shell
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessEidCache(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:869: undefined reference to `otThreadGetEidCacheEntry'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::Interpreter(ot::Instance*)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:292: undefined reference to `otThreadSetReceiveDiagnosticGetCallback'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessPSKc(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:1283: undefined reference to `otThreadGetPSKc'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:1297: undefined reference to `otThreadSetPSKc'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessRouterRole(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2389: undefined reference to `otThreadIsRouterRoleEnabled'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2404: undefined reference to `otThreadSetRouterRoleEnabled'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessState(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2577: undefined reference to `otThreadBecomeRouter'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2581: undefined reference to `otThreadBecomeLeader'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessChild(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:511: undefined reference to `otThreadGetChildInfoByIndex'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:556: undefined reference to `otThreadGetChildInfoById'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessChildMax(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:609: undefined reference to `otThreadGetMaxAllowedChildren'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:614: undefined reference to `otThreadSetMaxAllowedChildren'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessContextIdReuseDelay(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:660: undefined reference to `otThreadGetContextIdReuseDelay'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:665: undefined reference to `otThreadSetContextIdReuseDelay'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessLeaderWeight(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:1263: undefined reference to `otThreadGetLocalLeaderWeight'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:1268: undefined reference to `otThreadSetLocalLeaderWeight'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessNetworkIdTimeout(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:1544: undefined reference to `otThreadGetNetworkIdTimeout'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:1549: undefined reference to `otThreadSetNetworkIdTimeout'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessParentPriority(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:1628: undefined reference to `otThreadGetParentPriority'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:1633: undefined reference to `otThreadSetParentPriority'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessReleaseRouterId(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2131: undefined reference to `otThreadReleaseRouterId'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessRouter(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2297: undefined reference to `otThreadGetRouterInfo'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2332: undefined reference to `otThreadGetRouterInfo'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessRouterDowngradeThreshold(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2371: undefined reference to `otThreadGetRouterDowngradeThreshold'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2376: undefined reference to `otThreadSetRouterDowngradeThreshold'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessRouterSelectionJitter(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2422: undefined reference to `otThreadGetRouterSelectionJitter'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2428: undefined reference to `otThreadSetRouterSelectionJitter'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessRouterUpgradeThreshold(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2442: undefined reference to `otThreadGetRouterUpgradeThreshold'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:2447: undefined reference to `otThreadSetRouterUpgradeThreshold'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessJoinerPort(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:3007: undefined reference to `otThreadGetJoinerUdpPort'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:3012: undefined reference to `otThreadSetJoinerUdpPort'
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): In function `ot::Cli::Interpreter::ProcessNetworkDiagnostic(int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli.cpp:3476: undefined reference to `otThreadSendDiagnosticGet'
[... cut ...]
ext_proj/Install/ot/zephyr/lib/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli_dataset.o): In function `ot::Cli::Dataset::ProcessMgmtGetCommand(otInstance*, int, char**)':
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli_dataset.cpp:657: undefined reference to `otDatasetSendMgmtActiveGet'
/home/ramonribeiro/zephyr/samples/net/echo_client/build/zephyr/ext_proj/Source/ot/src/cli/cli_dataset.cpp:662: undefined reference to `otDatasetSendMgmtPendingGet'
collect2: error: ld returned 1 exit status
```

nashif, 2018-07-17:
issue fixed and merged

closed 2018-07-17 by nashif

Yug11, 2020-02-19:
Hi @ramonhpr,
I am a Newbie working on a project where I want to use Zephyr along with openThread and I don't know how exactly to use zephyr to build an FTD device .
could you please help me with running Zephyr as an FTD on two NRF52840 DK boards.

ramonhpr, 2020-02-27:
Sure. When I used zephyr, the FTD option was the default if you'd like to change the option, you should change that. (I used to run the command `make menuconfig` and search for the option MTD, but I think that now you can use the command `west` for that).

Since you are getting started in the zephyr I recommend you to follow the [getting started](https://docs.zephyrproject.org/latest/getting_started/index.html#build-the-blinky-application).

And for openthread you can use the samples [echo server](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_server/README.html#sockets-echo-server-sample) and [echo client](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_client/README.html).

Sorry for the late answer, I hope that I could help you.