zephyrproject-rtos/zephyr #8851: Allow creation of Zephyr libraries outside of Zephyr tree.

pdunaj, 2018-07-11:
At the moment if Zephyr library is defined outside of Zephyr tree (i.e. it is defined from application cmake files), build system rejects to build and link it.

This is due to the fact that Zephyr collects all libraries defined within its tree and does magic on them before leaving boilerplate.cmake. When library is defined after this part is executed build system will not see this library and so ignore it completely.
Note that this does not happen for zephyr sources.

Could build system allow adding out-of-tree zephyr libraries?

pdunaj, 2018-07-11:
@SebastianBoe

carlescufi, 2018-07-12:
cc @tejlmand

tejlmand, 2018-07-13:
Could you elaborate on what you exactly try to achieve ?

Are you creating a library, by which Zephyr itself depends upon, or is it a library that your app is depending upon?

Cause in the later case, it is important to have something depending on your library, e.g.:
`zephyr_library_named(MyLib)`
`zephyr_library_sources(<sources> )`
`target_link_libraries(app MyLib) # This line is important to specify MyLib is used by app`

or using plain CMake:
`add_library(MyLib STATIC <sources> )                                                                                                 `
`target_link_libraries(MyLib zephyr_interface) # To obtain zephyr list of compile flags`
`target_link_libraries(app MyLib)`

pdunaj, 2018-07-13:
@tejlmand Sure.
At the moment in our organization we use two repositories. One is a fork of Zephyr and a second is a repository with code specific to Nordic.
Some of the code in the second repository expands Zephyr itself and so should be linked with zephyr (not the app library). 

As I mentioned above this does not work at the moment as any library linked with zephyr must be defined as part of Zephyr build tree (i.e. cmake adding a link must be imported from main Zephyr cmake file).

The above limits Zephyr as with this approach application developer will not be able to provide out of tree extensions to the system. The only thing that could be done is to link with app library but I don't think the end effect effect would be the same. (or would it?)

SebastianBoe, 2018-08-02:
I can confirm that it is not possible to create a true out-of-tree Zephyr CMake library today.

There are several ways we could add support for this.

Allow users to declare Zephyr CMake libraries before boilerplate.cmake.

Re-write the "magic" in the core build scripts that depend on having a complete list of Zephyr CMake libraries to use generator expressions instead.

Introduce a footer and header boilerplate.cmake to allow the user to inject code at the appropriate time.

Allow custom 'app' hook build scripts that run in "kernel mode" and can therefore declare Zephyr CMake Libraries.

I would like to first investigate the feasibility of extracting the list of Zephyr CMake libraries with a generator expression. In the mean time Zephyr CMake libraries will either have to be maintained in-source or emulated through appropriate build scripts in the 'app'.

SebastianBoe, 2018-09-06:
This is blocked by https://github.com/zephyrproject-rtos/zephyr/pull/9562

SebastianBoe, 2018-12-05:
Still blocked by https://github.com/zephyrproject-rtos/zephyr/pull/9562

carlescufi, 2019-02-14:
@SebastianBoe is this still blocked? Can you update the status?

SebastianBoe, 2019-02-14:
The status is essentially the same. Except that the earlier proposed solutions (#9562 and #9914 ) have bit rotted and need rework.

SebastianBoe, 2019-08-09:
This solution proposal has now implemented:

`Allow custom 'app' hook build scripts that run in "kernel mode" and can therefore declare Zephyr CMake Libraries.`

The hook is ZEPHYR_EXTRA_MODULES.

App developers can now do

```
list(APPEND ZEPHYR_EXTRA_MODULES /path/to/my/external/library)

cmake_minimum_required(VERSION 3.13.1)

include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
```

and create zephyr libraries outside of the zephyr tree.

closed 2019-08-09 by SebastianBoe

marc-hb, 2019-08-19:
> This solution proposal has now implemented:

Care to provide direct link(s) to the corresponding PR(s)?