zephyrproject-rtos/zephyr #8864: ARM MPU allowing reads to kernel memory

andrewboie, 2018-07-11:
Reproducible with this PR:
https://github.com/zephyrproject-rtos/zephyr/pull/8863
Run tests/kernel/mem_protect/syscalls on sam_e70_xplained.

The check to _arch_buffer_validate on a buffer defined with __kernel is not failing as it ought to be.

Not reproducible on frdm_k64f, so this seems specific to ARM MPU.

andrewboie, 2018-07-11:
The MPU *is* protecting the memory if I call libc strlen on the same string, so this is definitely a problem with the validation function and not the MPU configuration itself.

ioannisg, 2018-07-12:
@andrewboie submitted a PR that fixes this; the test you indicate is now successful after re-basing PR over #8868 .

closed 2018-07-12 by MaureenHelm

andrewboie, 2018-07-12:
thanks so much!!

PR #8868 (2018-07-12): arch: arm: mpu: fix _get_region_ap(.) function
This commit fixes a bug in _get_region_ap(.) by adding
parentheses to enforce the correct operator precedence.

Fixes #8864.
commit 8bfddb52 (2018-07-12): arch: arm: mpu: fix _get_region_ap(.) function

This commit fixes a bug in _get_region_ap(.) by adding
parentheses to enforce the correct operator precedence.

Fixes #8864.