zephyrproject-rtos/zephyr #8871: drivers: can: Compiling error due to stm23Cube update

alexanderwachter, 2018-07-12:
The stm32 can driver does not compile for stm32L4 series after the stm32Cube update.
This is because names have been changed. I already created a fix but this will brake the driver for stm32F0 series because they are on an older version of the stm32Cube.

alexanderwachter, 2018-07-12:
@erwango are you going to update the stm32Cube for the other stm32 series too? Otherwise I have to remove the HAL_CAN_Init function from the zephyr driver.

erwango, 2018-07-30:
@alexanderwachter, indeed, my bad, I didn't pay enough attention to this API breakage.
I did the upgrade of Cube packages for all the packages available the day I did it. So, today, there are no more recent version of cube packages available. I don't have information right now on the Cube CAN API roadmap, I guess if all series should be updated this might take some months.

One solution to this issue on impacted series (F4, F7 and L4), is to get back on legacy CAN API:
-disable HAL_CAN_MODULE_ENABLED 
-enable HAL_CAN_LEGACY_MODULE_ENABLED

Impact is located in following files: 
-ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_conf.h
-ext/hal/st/stm32cube/stm32f7xx/drivers/include/stm32f7xx_hal_conf.h
-ext/hal/st/stm32cube/stm32l4xx/drivers/include/stm32l4xx_hal_conf.h

Can you have a try with this fix?
https://github.com/erwango/zephyr/tree/fix_can_legacy_api

nashif, 2018-08-14:
@erwango any update on this

alexanderwachter, 2018-08-25:
@erwango sorry for this late response but I was on holiday for about one month :-)
Thanks for switching back to legacy API. I think we will switch to the new API as soon as
available.