zephyrproject-rtos/zephyr #8872: New shell subsystem

nordic-krch, 2018-07-12:
New shell subsystem will be ported from nRF5 SDK (known there as nrf_cli). It has following features:
- multi-instance
- user friendly features: autocompletion, wildcards, coloring, metakeys (arrows, backspace, ctrl+u, etc.), history
- integration with logger subsystem - logs are multiplexed with data outputed by the shell ensuring that they do not interfere
- shell transport generic api - indenpendent of the shell
- static commend tree (root command links to subcommands, subcommands can link to another subcommands set)
- dynamic subcommands - autocompletion can use dynamic data not known during compilation

New shell subsystem shall be merged when legacy shell is totally replaced targeting 1.14 release.

nordic-krch, 2018-07-12:
fyi: @carlescufi @jarz-nordic @pizi-nordic @nashif

pfalcon, 2018-07-12:
I'm all for detailed bug tickets, but her were have ~20 *absolutely identical* tickets of "Adapt xxx to new shell subsystem". What's the purpose of them? Why just having checkboxes in this master ticket isn't enough?

nordic-krch, 2018-07-17:
@pfalcon I agree that this may not look nice but introducing this new console will require adaption work which will be split across multiple contributors. Separate tasks can help in terms of taking responsibility.

koffes, 2018-09-13:
Looking forward to these new features. Will multiple transports be supported so that SYS_LOG may be routed to RTT ch. 0 and the shell be accessible over UART or RTT ch. 1 at the same time?

jarz-nordic, 2018-09-13:
@koffes yes it will be possible. But why would you like to have separate channel for SYS_LOG and shell? New shell will allow you to have SYS_LOG and shell on one RTT channel.

koffes, 2018-09-13:
@jarz-nordic During debugging, we often have significant SYS_LOG output, whilst we want to be able to use the shell. Reading from shell/writing to shell in the middle of all this log "noise" is difficult.

jarz-nordic, 2018-09-19:
@koffes : New shell is in upstream master.

jarz-nordic, 2018-10-17:
All examples have been migrated.

closed 2018-10-17 by jarz-nordic