zephyrproject-rtos/zephyr #8898: Failed test: kernel.timer.timer_periodicity on nrf51

zephyrbot, 2018-07-12:
Test tests/kernel/timer/timer_api/kernel.timer failing on nrf51:

```
***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.12.99 (delayed boot 1000ms) *****
 Running test suite timer_api
 ===================================================================
 starting test - test_timer_duration_period
 PASS - test_timer_duration_period
 ===================================================================
 starting test - test_timer_period_0
 PASS - test_timer_period_0
 ===================================================================
 starting test - test_timer_expirefn_null
 PASS - test_timer_expirefn_null
 ===================================================================
 starting test - test_timer_periodicity

     Assertion failed at zephyr.git/tests/kernel/timer/timer_api/src/main.c:171: test_timer_periodicity: (((delta) >= (50)) && ((delta) <= (50) + (1))) is false

 FAIL - test_timer_periodicity
 ===================================================================
 starting test - test_timer_status_get
 PASS - test_timer_status_get
 ===================================================================
 starting test - test_timer_status_get_anytime
 PASS - test_timer_status_get_anytime
 ===================================================================
 starting test - test_timer_status_sync

     Assertion failed at zephyr.git/tests/kernel/timer/timer_api/src/main.c:53: duration_expire: k_uptime_delta(&tdata.timestamp) >= 50 is false

 ASSERTION FAIL [!(thread->base.user_options & (1 << 0))] @ zephyr.git/arch/arm/core/thread_abort.c:37:
 	essential thread aborted

```

nashif, 2018-07-12:
also failed on nrf52840_pca10056

spoorthik, 2018-08-16:
With the commit: 734eda39f5868207bb6a9d0eba346d24e153eba3 subtest test_timer_period_0 is also failing on NRF51

```
Running test suite timer_api
===================================================================
starting test - test_timer_duration_period
PASS - test_timer_duration_period
===================================================================
starting test - test_timer_period_0

    Assertion failed at /home/kspoorth/work/latest_zephyr/tests/kernel/timer/timer_api/src/main.c:159: test_timer_period_0: tdata.expire_cnt == 1 is false

FAIL - test_timer_period_0
===================================================================
starting test - test_timer_expirefn_null
PASS - test_timer_expirefn_null
===================================================================
starting test - test_timer_periodicity

    Assertion failed at /home/kspoorth/work/latest_zephyr/tests/kernel/timer/timer_api/src/main.c:240: test_timer_periodicity: (((delta) >= (50)) && ((delta) <= (50) + (1))) is false

FAIL - test_timer_periodicity
===================================================================
starting test - test_timer_status_get
PASS - test_timer_status_get
===================================================================
starting test - test_timer_status_get_anytime

    Assertion failed at /home/kspoorth/work/latest_zephyr/tests/kernel/timer/timer_api/src/main.c:299: test_timer_status_get_anytime: k_timer_status_get(&timer) == 4 is false

FAIL - test_timer_status_get_anytime
===================================================================
starting test - test_timer_status_sync
PASS - test_timer_status_sync
===================================================================
starting test - test_timer_k_define
PASS - test_timer_k_define
===================================================================
starting test - test_timer_user_data
PASS - test_timer_user_data
===================================================================
===================================================================
PROJECT EXECUTION FAILED

```

PR #9745 (2018-08-31): kernel: Improve precision of ticks and ms conversions
The following 2 improvements are contained in this patch:

- When converting from ms to ticks, instead of using hardware cycles
  per tick, use hardware cycles per second. This ensures that the
  multiplication is done before the division, increasing precision.
- When converting from ticks to ms, instead of using cycles per tick
  and cycles per sec, use ticks per sec. This too increases the
  precision.

The concept is to make the dividend as large as possible compared to the
divisor in order to lose as little precision as possible.

Fixes [... cut]
commit c7d27344 (2018-08-31): kernel: Improve precision of ticks and ms conversions

The following 2 improvements are contained in this patch:

- When converting from ms to ticks, instead of using hardware cycles
  per tick, use hardware cycles per second. This ensures that the
  multiplication is done before the division, increasing precision.
- When converting from ticks to ms, instead of using cycles per tick
  and cycles per sec, use ticks per sec. This too increases the
  precision.

The concept is to make the dividend as large as possible compared to the
divisor in order to lose as little precision as possible.

Fixe [... cut]
closed 2018-08-31 by nashif