zephyrproject-rtos/zephyr #8921: remove -Wno-pointer-sign

nashif, 2018-07-13:
Remove `zephyr_cc_option(-Wno-pointer-sign)` and fix all code that mixes signed/unsigned pointers.

psarvesh, 2024-02-10:
Locally, to try to fix this, I removed the flag from cmake files and then ran twister. I see a bunch of build failures related to signed-ness.

Should I create a PR with all of those corrected?

averagewagon, 2024-05-20:
There are a few hundred files containing pointer sign issues, which will all need to be resolved to remove `-Wno-pointer-sign`. Considering that a PR which resolves these issues will touch source files across the entire Zephyr codebase, should this issue be marked as "treewide"?

https://docs.zephyrproject.org/latest/contribute/guidelines.html#treewide-changes

n-eq, 2025-02-21:
> Locally, to try to fix this, I removed the flag from cmake files and then ran twister. I see a bunch of build failures related to signed-ness.
> 
> Should I create a PR with all of those corrected?

Hi @psarvesh , did you open a PR? do you want help with this?

ritesh006, 2025-09-01:
Hi @nashif can I take this issue?