zephyrproject-rtos/zephyr #8922: Enhance 3rd party libc support

nashif, 2018-07-13:
Using newlib by default across the project is currently not possible due to type conflicts. Zephyr should build with the newlib that comes with the Zephyr SDK and any newlib that comes pre-integrated with 3rd party toolchains.

Beside newlib, we need to support any 3rd libc and not limit the support to newlib and the minimal libc in the zephyr tree. A few options were discussed and being explored:
- make newlib zephyr aware
- Use PRI macro for 32bit integer types 
- Make types in Zephyr compatible with newlib
- etc.

A few bugs are open that are related to this issue:

-

nashif, 2018-07-16:
@galak @carlescufi @dleach02 FYI

nashif, 2018-07-16:
I have just found what appears like a bug in our Zephyr SDK configuration. The toolchain does not expose any headers like unistd.h or sys/types.h, unlike any other toolchain we use. for example when you build with xtools or arm embedded toolchain, all the headers are available to zephyr, not with the Zephyr SDK forcing us to add all of those headers into the minimal libc in a non-compatible way. 

This PR #8944 add the missing include

nashif, 2018-07-19:
more issues #7441

cfriedt, 2022-03-19:
@nashif - This can probably be closed now, no?

stephanosio, 2022-10-04:
> This can probably be closed now, no?

@cfriedt technically, this issue is not resolved.

> make newlib zephyr aware

https://github.com/zephyrproject-rtos/sdk-ng/issues/350

> Use PRI macro for 32bit integer types
> Make types in Zephyr compatible with newlib

https://github.com/zephyrproject-rtos/zephyr/issues/46032

stephanosio, 2024-02-08:
Closing since Zephyr is currently compatible with various third-party C libraries without any major issues.

closed 2024-02-08 by stephanosio