zephyrproject-rtos/zephyr #8931: STM32F4 CAN sample project does not compile

markusr, 2018-07-13:
I am trying to get CANbus sample to run. However, I cannot compile the sample program. I am working with NUCLEO-L432KC board.

```
zephyr/samples/drivers/CAN$ mkdir build && cd build
$ cmake -GNinja -DBOARD=nucleo_l432kc ..
--> runs fine
```

Then I want to compile ```ninja``` which fails:
```
/home/markus/Entwicklung/tmp/zephyr-zephyr-v1.12.0/drivers/can/stm32_can.c:795:24: error: ‘CONFIG_CAN_1_BASE_ADDRESS’ undeclared here (not in a function)
  .can = (CAN_TypeDef *)CONFIG_CAN_1_BASE_ADDRESS,
```

See full log [here](https://pastebin.com/rwJFHAKR)

markusr, 2018-07-16:
I did some more investigation. It seems that ST has changed the HAL of the CAN driver:
https://community.st.com/thread/47095-stm32f4xx-rework-of-hal-can-driver-examples

@alexanderwachter Can you take a look?

alexanderwachter, 2018-07-16:
Hello @markusr,

I already opened an issue (Issue #8871) for that. I'm sorry for that but at the moment I can only help you with a quick and dirty solution.
https://github.com/alexanderwachter/zephyr/tree/can_stmcube_1_12
Currently my plan is to remove the stm32Cube HAL from the can_stm32 driver, but this still in progress because I will not be available for 4 weeks from now.
Actually you can take a look at this version but the time out for init mode is still missing and it's not tested. 
https://github.com/alexanderwachter/zephyr/tree/can_replace_hal_init

markusr, 2018-07-16:
@alexanderwachter your tree `can_stmcube_1_12` compiles fine. I will try to connect the CAN interface to real hardware and test the code.

alexanderwachter, 2018-07-16:
@markusr thanks for testing!
Be aware that this code does not work with the stm32f0 series because the stmCube for those has an
incompatible version. I think the final version will be the one without stm32Cube HAL_CAN_Init.
If you need the possibility to recover from bus-off or read the actual bus state, pleas have a look at
https://github.com/alexanderwachter/zephyr/tree/can_extend_api
I will prepare a PR for this after my holidays.

alexanderwachter, 2018-07-16:
@markusr note that NUCLEO-L432KC does not have the external clock enabled by default.
With CAN its highly recommended to use the 8MHz Crystal stabilized reference clock from the stlink.

markusr, 2018-07-16:
@alexanderwachter 
Thank you for all these informations.

> note that NUCLEO-L432KC does not have the external clock enabled by default.

You are correct. The external clock input is connected to the MCO pin of the header (You have to add/remove some solder bridges to use it.).

> With CAN its highly recommended to use the 8MHz Crystal stabilized reference clock from the stlink.

Where did you get this information from? Testing? I cannot find anything in the [data sheet for L432KC](https://www.st.com/resource/en/datasheet/stm32l432kb.pdf)

Only thing I found was in [user manual](https://www.st.com/resource/en/user_manual/dm00231744.pdf)

> 6.8 OSC clock
>
>In applications where VCP is used for communication at a speed higher than 9600 bauds, it may be
>needed to use this solder bridge configuration, to use 8 MHz clock (MCO from ST-LINK) and get a more precise frequency.

This, however, refers to the VCP (virtual COM port) and not the CAN interface.

alexanderwachter, 2018-07-16:
> Where did you get this information from? Testing? I cannot find anything in the data sheet for L432KC

Testing and ISO 12.4.2.5 defines that df <= min(phase_seg1, phase_seg2)/(2 * (13 * t_bit - phase_seg2)) and df <= SJW/(20 * t_bit) and the maximum difference of two oscillators shall be 2 * df * f_nom. Usually this you cannot guarantee with an internal osc over temperature.

PR #9184 (2018-08-14): STM32: CAN: Enable legacy CAN API (F4, F7, L4)
In recent version of STM32Cube packages for F4, F7 and L4 series, a new CAN API was introduced that is not compatible with earlier API. This change breaks the current STM32 CAN driver.

Disable the new CAN API and enable the legacy API.

Fixes #8931
commit ee29e603 (2018-08-14): ext/hal: stm32cube: STM32L4: Enable legacy CAN API

A new CAN API has been delivered on recent L4 stm32cube.
This new API breaks current CAN driver. Disable the new API and
enable the legacy API.

Fixes #8931
closed 2018-08-14 by nashif