zephyrproject-rtos/zephyr #8933: doc: build WARNING on windows 7

hongshui3000, 2018-07-14:
MSG:
````
D:\zephyr\doc\api\file_system.rst:20: WARNING: Duplicate declaration.
D:\zephyr\doc\api\io_interfaces.rst:24: WARNING: Duplicate declaration.
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected end of definition. [error at 8]
  in6_addr.in6_u
  --------^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected end of definition. [error at 7]
  in_addr.in4_u
  -------^
D:\zephyr\doc\api\networking.rst:72: WARNING: Invalid definition: Expected end of definition. [error at 11]
  net_context.options
  -----------^
D:\zephyr\doc\api\networking.rst:90: WARNING: Invalid definition: Expected end of definition. [error at 12]
  net_stats_tc.sent
  ------------^
D:\zephyr\doc\api\networking.rst:90: WARNING: Invalid definition: Expected end of definition. [error at 12]
  net_stats_tc.recv
  ------------^

D:\zephyr\doc\subsystems\sensor.rst:37: WARNING: Duplicate declaration.
D:\zephyr\doc\api\file_system.rst:20: WARNING: Duplicate declaration.
D:\zephyr\doc\api\io_interfaces.rst:24: WARNING: Duplicate declaration.
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected end of definition. [error at 8]
  in6_addr.in6_u
  --------^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected end of definition. [error at 7]
  in_addr.in4_u
  -------^
D:\zephyr\doc\api\networking.rst:72: WARNING: Invalid definition: Expected end of definition. [error at 11]
  net_context.options
  -----------^
D:\zephyr\doc\api\networking.rst:90: WARNING: Invalid definition: Expected end of definition. [error at 12]
  net_stats_tc.sent
  ------------^
D:\zephyr\doc\api\networking.rst:90: WARNING: Invalid definition: Expected end of definition. [error at 12]
  net_stats_tc.recv
  ------------^

New errors/warnings found, please fix them:

D:\zephyr\doc\api\file_system.rst:20: WARNING: Duplicate declaration.

D:\zephyr\doc\api\io_interfaces.rst:24: WARNING: Duplicate declaration.

D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^

D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^

D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^

D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected end of definition. [error at 8]

  in6_addr.in6_u

  --------^

D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^

D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^

D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^

D:\zephyr\doc\api\networking.rst:42: WARNING: Invalid definition: Expected end of definition. [error at 7]

  in_addr.in4_u

  -------^

D:\zephyr\doc\api\networking.rst:72: WARNING: Invalid definition: Expected end of definition. [error at 11]

  net_context.options

  -----------^

D:\zephyr\doc\api\networking.rst:90: WARNING: Invalid definition: Expected end of definition. [error at 12]

  net_stats_tc.sent

  ------------^

D:\zephyr\doc\api\networking.rst:90: WARNING: Invalid definition: Expected end of definition. [error at 12]

  net_stats_tc.recv

  ------------^

D:\zephyr\doc\subsystems\sensor.rst:37: WARNING: Duplicate declaration.
```````

carlescufi, 2018-07-14:
@hongshui3000 this should have been fixed by https://github.com/zephyrproject-rtos/zephyr/commit/5e5fe0d74f4be756af268689f6c645884fb70dd3. Can you make sure you have this commit?

carlescufi, 2018-07-16:
@hongshui3000 can you please let me know if the commit above solves this for you?

hongshui3000, 2018-07-16:
thanks .   
solved

closed 2018-07-16 by hongshui3000