zephyrproject-rtos/zephyr #8958: Bluetooth: Proprietary vendor specific opcode discovery

cvinayak, 2018-07-17:
There are Zephyr vendor specific opcode, there will be Zephyr manufacturer vendor specific opcode, Android Bluetooth HCI requirements opcode [1], and Microsoft-defined Bluetooth HCI extensions [2]

Zephyr BLE controller implementation based vendors implementing proprietary features will reserve a range of command opcode in the vendor specific HCI group (OGF 0x3F) at build time. For the scalability of Zephyr specified vendor specific command opcodes, it is necessary to resolve undesirable overlap of conflicting command opcodes values between the host and controller when building new versions, this probably addressed already in Zephyr Read Host Stack Commands Command [3].

As a BLE controller implementation, I want to implement new Zephyr defined vendor HCI command opcodes by moving my overlapping vendor specific proprietary command opcode values away and my existing host be able to discover this change and continue to use the old zephyr and proprietary features as before.

Compatibility requirement (where M > N):
vN.0 host, vN.0 controller --> vN.0 Zephyr VS and Proprietary VS available.
vN.0 host, vM.0 controller --> vN.0 Zephyr VS and Proprietary VS available.
vM.0 host, vN.0 controller --> vN.0 Zephyr VS and Proprietary VS available.
vM.0 host, vM.0 controller --> vM.0 Zephyr VS and Proprietary VS available.

Proposal:
Update the Zephyr Read Host Stack Commands Command to return a 16-bit Vendor_ID[i] value of 0x0000 (reserved) and 10-bit Opcode_Base[i] of 0xXXXX which will convey the controller manufacturer's vendor specific opcode base.

[1] - https://source.android.com/devices/bluetooth/Android-6.0-Bluetooth-HCI-Reqs.pdf
[2] - https://docs.microsoft.com/en-us/windows-hardware/drivers/bluetooth/microsoft-defined-bluetooth-hci-commands-and-events
[3] - https://github.com/zephyrproject-rtos/zephyr/blob/16a28bf10689d8ea3b2700863adb8284264d11af/doc/subsystems/bluetooth/hci.txt#L699

aescolar, 2018-07-17:
CC: @pskhansen @mped-oticon

carlescufi, 2018-07-17:
CC @joerchan

jhedberg, 2018-08-14:
> Proposal:
> Update the Zephyr Read Host Stack Commands Command to return a 16-bit Vendor_ID[i] value of 0x0000 (reserved) and 10-bit Opcode_Base[i] of 0xXXXX which will convey the controller manufacturer's vendor specific opcode base.

Maybe I'm not understanding something here, but why would 0x0000 be used instead of the controller manufacturer's own company ID?

cvinayak, 2018-08-14:
> Maybe I'm not understanding something here, but why would 0x0000 be used instead of the controller manufacturer's own company ID?

These are private (proprietary) opcodes, where proprietary host (implementation) commands interface with these proprietary features in the controller. No other vendor's host implementation will know about these features (commands). There is no necessity to update/add vendor id (enum value) to the hci.txt file.

The vendor can decide to define public vendor specific commands, for which he can have an entry with his vendor id in the array. In this case, other host implementations (say, for instance Bluez) can use the features, like the Android or Microsoft specific commands. In this case, the vendor id enum value is to be added in the hci.txt.

carlescufi, 2018-08-16:
@holtmann can you take a look at this one?

cvinayak, 2021-05-06:
Closing this as there is no traction for sometime now.

closed 2021-05-06 by cvinayak