zephyrproject-rtos/zephyr #8960: Tcp connection not connecting

fadhel086, 2018-07-17:
Hi,
I've been developing Ethernet drivers for qemu_cortex_m3 (stellaris lm3s6965).

For testing the driver with qemu, a virtual network was made with vde_switch. One qemu-arm runs zephyr with Ethernet driver and another qemu-x86_64 with linux.
We could successfully test using ping commands and UDP. In UDP zephyr acts as the client and linux machine as server.
For a TCP test, where zephyr acts as the client and linux machine as server, there is no response coming from the server.
tcpdump showed as only SYN packets. No other response was coming from the server.

When tested two different qemu-linux machines for TCP, the system is responding.

Would like to know anything on above, or previous threads on it.

regards

fadhel086, 2018-07-17:
Using the latest zephyr v1.12

jukkar, 2018-07-17:
Can you make sure that your linux host does not block the packets? In my setup, I had similar issues as my linux firewall was blocking incoming connections from zephyr.

fadhel086, 2018-07-18:
I shall, check on that!

fadhel086, 2018-07-19:
Hi., Got the issue solved. 
Had a bug in driver and few config issues

Thanx

closed 2018-07-19 by fadhel086