zephyrproject-rtos/zephyr #8965: Kernel API is using explicit concatenation (##) which has limits

nordic-krch, 2018-07-17:
https://github.com/zephyrproject-rtos/zephyr/blob/a07d98a62fff0330fdb42d24e139c613a20c4905/include/kernel.h#L3288

Explicit use of token pasting leads to issues when provided parameter contains macros. Macros are not resolved then.

Unless there are reasons behind using explicit concatenation, it should be replaced by UTIL_CAT (https://github.com/zephyrproject-rtos/zephyr/blob/a07d98a62fff0330fdb42d24e139c613a20c4905/include/misc/util.h#L198) instead.

andyross, 2018-07-17:
I mean, that's true... but the q_name parameter in this case is a symbol name provided as a literal to a declaration wrapper.  It doesn't make sense to make it recursively expand as a macro.  Do we really care?  And UTIL_CAT isn't really all that clear vs. natural language syntax, IMHO, but that's a taste thing.

Not that I'd -1 a patch that did this, but do we really want to bother for usage like this?

nordic-krch, 2018-08-01:
The problem that I found was that once UTIL_CAT() is used as one of the argument of ## compilation fails.

Here is the use case I've seen before:

Suppose that you have a module which has own MODULE_DEFINE(_name) and uses kernel module, like:

```
#define MODULE_DEFINE(_name) \
K_MSGQ_DEFINE(_name##_msgq,...)
struct module _name __attribute__ ((section("." STRINGIFY(_name))))= {
.msgq = &_name##_msgq
}
```
Compilation fails on that call:
` MODULE_DEFINE(UTIL_CAT(my_module, MODULE_PRIORITY_HIGH))`

In that case I used define in the name to enforce sorting by priority in given memory section used for that modules.

So, imo it is better to use UTIL_CAT (maybe it's better to rename it to CONCAT_2, CONCAT_3, etc.) to not limit the user.