zephyrproject-rtos/zephyr #8976: nordic: watchdog: Cannot be initialized - circular dependency

Olivier-ProGlove, 2018-07-17:
I was looking at adding watchdog support in my Zephyr application. I was following Zephyr [watchdog test](https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c) as [the watchdog sample uses the deprecated API](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/drivers/watchdog/src/main.c).

The initialization code is pretty simple:

```
static int app_watchdog_init(void) {
	int err;
	struct device *wdt = device_get_binding(CONFIG_WDT_0_NAME);
	if (wdt == NULL) {
		printk("Could not find watchdog device %s\n", CONFIG_WDT_0_NAME);
		return -ENODEV;
	}

	watchdog_cfg.callback = NULL;
	watchdog_cfg.flags = WDT_FLAG_RESET_SOC;
	watchdog_cfg.window.max = 2000;
	err = wdt_install_timeout(wdt, &watchdog_cfg);
	if (err < 0) {
		return err;
	}

	err = wdt_setup(wdt, 0);
	if (err < 0) {
		return err;
	}
	return 0;
}
```

The issue is using this code, I have the following assert in Nordic HAL Watchdog driver:

```
assertion "m_state == NRFX_DRV_STATE_INITIALIZED" failed: file "/tmp/zephyr/ext/hal/nordic/nrfx/drivers/src/nrfx_wdt.c", line 119, function: nrfx_wdt_channel_alloc
```
The assert occurs in `wdt_install_timeout()`. It expects the HAL function `nrfx_wdt_init()` being invoked before `nrfx_wdt_channel_alloc()`.

If I call `wdt_setup()` before calling `wdt_install_timeout()` I have a different assert:

```
assertion "m_alloc_index != 0" failed: file "/tmp/zephyr/ext/hal/nordic/nrfx/drivers/src/nrfx_wdt.c", line 97, function: nrfx_wdt_enable
```

Again, it expects a watchdog channel to be allocated. But it is only done by `wdt_install_timeout()`.

One possible fix would be to call `nrfx_wdt_init()`  in `wdt_install_timeout()` as Zephyr Watchdog API seems to require `wdt_install_timeout()` to be called before `wdt_setup()`.
And Nordic Watchdog driver only supports one channel: https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/watchdog/wdt_nrfx.c#L126. So `wdt_install_timeout()` should only be called once.

cc: @kl-cruz

kl-cruz, 2018-07-18:
Hi, 
You are right. Thanks for this!
This possible fix could work correctly, but we can use channels implicitely using mechanism similar to channels called reload request (we have to feed each configured reload request register in wdt to prevent mcu reset). 
So... as you mentioned it will work only with first allocation (one channel).
I looked deeply into this issue and I have idea how to fix it. I will prepare pull-request asap.

Olivier-ProGlove, 2018-07-30:
I confirm @kl-cruz'spull-request fix the issue.

closed 2018-07-30 by Olivier-ProGlove