zephyrproject-rtos/zephyr #8979: Failed test: usb.device.dfu.bank_erase

zephyrbot, 2018-07-18:
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.12.0-831-g5b8e4ae (delayed boot 1000ms) *****
Running test suite test_mcuboot_interface
===================================================================
starting test - test_bank_erase

    Assertion failed at zephyr/tests/subsys/dfu/mcuboot/src/main.c:59: test_bank_erase: (pass)
fail
FAIL - test_bank_erase
===================================================================
starting test - test_request_upgrade
PASS - test_request_upgrade
===================================================================
starting test - test_write_confirm
PASS - test_write_confirm
===================================================================
===================================================================
RunID: :pmb1
PROJECT EXECUTION FAILED

jfischer-phytec-iot, 2018-07-18:
Known issue, also reported in #8734

nvlsianpu, 2018-09-10:
@jfischer-phytec-iot This issue is something completely different than #8734 (a timing issue) you mentioned.

jfischer-phytec-iot, 2018-09-10:
@nvlsianpu Looks like the title is wrong "Failed test: usb.device.dfu.bank_erase on nrf52840_pca10056"

carlescufi, 2018-12-13:
Closing this since it's a result of flash wear

closed 2018-12-13 by carlescufi