zephyrproject-rtos/zephyr #8982: tests/drivers/watchdog/wdt_basic_api results in FATAL EXCEPTION on esp32

pswarnak, 2018-07-18:
The test **tests/drivers/watchdog/wdt_basic_api** results in FATAL EXCEPTION.

**Arch:** xtensa
**Board:** esp32
**Zephyr commit:** 5b8e4ae4df378d0da32ce36266246fb530a6309d.

**Error Console Log:**
```
***** Booting Zephyr OS v1.12.0-831-g5b8e4ae (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
 ** FATAL EXCEPTION
 ** CPU 0 EXCCAUSE 20 PS 0x00060120 PC 0x00000000 VADDR 0x00000000
 **  A0 0x80082419  SP 0x3ffb1b70  A2 0x00000000  A3 0x00000000
 **  A4 0x3ffb0000  A5 0x3ffb0208  A6 0x3ffb0208  A7 0x00000000
 **  A8 0x8008099f  A9 0x3ffb1b20 A10 0x3ffb0000 A11 0x3ffb1280
 ** A12 0x3ffb036f A13 0x3ffb26e0 A14 0x3ffb021c A15 0x3ffb14c0
 ** LBEG 0x00000000 LEND 0x00000000 LCOUNT 0x00000000
 ** SAR 0x00000000
Current thread ID = 0x3ffb021c
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
```

**Steps to reproduce:**
```
cd zephyr/tests/drivers/watchdog/wdt_basic_api/
rm -rf build && mkdir build && cd build
cmake -D BOARD=esp32 -GNinja ../
ninja flash

Check the console log.
```

nniranjhana, 2018-07-19:
We already have bugs reported for this, so this maybe a duplicate.
#8222 (this was reported on ESP32 too but was closed as a duplicate for: #8097)

closed 2018-07-19 by nniranjhana

pswarnak, 2018-07-19:
yeah, i created this issue as it was on a different architecture with different error signature.
I updated the existing issue with esp32 details too. Thanks.