zephyrproject-rtos/zephyr #8988: [Coverity CID :187079] Integer handling issues in /subsys/net/l2/ethernet/gptp/gptp.c

mandarcthorat1, 2018-07-18:
Static code scan issues seen in File: /subsys/net/l2/ethernet/gptp/gptp.c
 Category: Integer handling issues
 Function: gptp_set_time_itv
 Component: Networking
 CID: 187079
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

mandarcthorat1, 2019-03-22:
The problem is detected again by latest coverity scan. Please have a look

jukkar, 2019-03-25:
What is the coverity scan id for this new issue? The original one 187079 points to code that is already fixed.

jukkar, 2019-03-25:
I wonder why coverity is reporting this now instead of immediately after the previous fix was reported. Anyway, investigating.

commit 27fef49d (2018-07-20): net: gptp: Check overflow of log msg interval

Make sure that we do not overflow when creating UScaledNS
value for interval.

Coverity-CID: 187079
Fixes #8988
PR #14878 (2019-03-25): net: gptp: Fix bit shifting in time interval
It is invalid to try to bit shift the same amount of bits as
what is the number of bits in the left expression’s type.

Coverity-CID: 187079
Fixes #8988
commit ac4db1b3 (2019-03-25): net: gptp: Fix bit shifting in time interval

It is invalid to try to bit shift the same amount of bits as
what is the number of bits in the left expression’s type.

Coverity-CID: 187079
Fixes #8988
closed 2019-03-25 by nashif