zephyrproject-rtos/zephyr #9015: eth_sam_gmac driver (and BOARD=sam_e70_xplained using it) sets unduly high memory requirements on the IP stack overall

pfalcon, 2018-07-18:
eth_sam_gmac is quite an advanced Ethernet driver, the only one using DMA, etc.

But at the same time it does something which no other Ethernet (or non-Ethernet) driver does: it enforces at compile-time (using #error directives) high limits on the number of fragment buffers used by the IP stack (CONFIG_NET_BUF_RX_COUNT, CONFIG_NET_BUF_TX_COUNT): https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/ethernet/eth_sam_gmac.c#L43 . This has the following negative effect on the entire IP stack and Zephyr overall:

**Tests/sample apps which perfectly build for any other board, break when building for boards using the eth_sam_gmac driver, e.g. BOARD=sam_e70_xplained**. As part of Zephyr CI, we build tests/samples for as many boards as possible, so this negatively affects Zephyr CI overall.

As a response to it, there are attempts to work around eth_sam_gmac's behavior by bumping CONFIG_NET_BUF_RX_COUNT, CONFIG_NET_BUF_TX_COUNT higher, e.g. https://github.com/zephyrproject-rtos/zephyr/commit/a0df4f66984b3cc52c07da52e74fdadf7c79b728, https://github.com/zephyrproject-rtos/zephyr/pull/8966, etc. (there were definitely more sam_e70_xplained specific commits like that). There're however following issues with such workarounds:

1. It negatively affects available memory on smaller boards. We probably didn't face cases of running out of RAM just yet, but can very easy start to see with more complex examples integrating other memory-hungry code (e.g. TLS).
2. It adversely affects IP stack performance/stability (which is due to bugs in the IP stack, but nonetheless, eth_sam_gmac's behavior complicates behavior here and now for many other boards). An example is  https://github.com/zephyrproject-rtos/zephyr/pull/7831 .

This ticket is created to capture the concern, and consider various ways out of the situation.

pfalcon, 2018-07-18:
@nashif, @jukkar : Let's take the discussion from https://github.com/zephyrproject-rtos/zephyr/pull/8966#discussion_r203013732 into this ticket.

jukkar, 2018-07-19:
I think the issue is not really in sam-e70 ethernet driver. It is just trying to set a lower limit of how many network buffers are configured so that the driver can receive a full ethernet frame. To me this sounds like a sane thing to do.

What is more questionable, is that various unit tests in tests/net are compiled and run in atmel board which then fails as there is not enough buffers. IMHO running these tests in this hw (or any other real hw) is not really needed as we are running these tests in qemu and native_posix which is enough.

mnkp, 2018-07-21:
Drivers like eth_dw or eth_mcux reserve a private RX buffer in RAM - one continuous chunk - large enough to hold at least one full Ethernet frame. Upon receiving a frame they copy it form this private RX buffer to the RX fragment buffers provided by the networking stack. E.g. eth_mcux.c reserves the private RX buffer on line 135
```
static u8_t __aligned(ENET_BUFF_ALIGNMENT)
rx_buffer[CONFIG_ETH_MCUX_RX_BUFFERS][ETH_MCUX_BUFFER_SIZE];
```
here by default RAM for two full Ethernet frames is reserved.

In contrary eth_sam_gmac is a zero-copy networking implementation of an Ethernet driver as documented in the file header. It has no private RX buffer and places all the incoming data directly in RX fragment buffers provided by the networking stack. Therefore its requirements on the amount of available RX fragment buffers are higher than for the other drivers. 

The minimum buffer requirements are ensured by the compile time checks, like the one on line 43. It is vital that the checks are active.

Since they were disabled for unit tests it resulted in the bug report: #8963.

mnkp, 2018-07-21:
Of course, just because eth_sam_gmac requires higher amount of RX buffers we should not force remaining drivers to do the same. Maybe sane values could be set by a SoC specific _defconfig file?

mnkp, 2018-08-07:
To resolve this issue we need to decide how to define a 'reasonable' amount of network buffers for a given Ethernet driver without forcing all the drivers to use the same values. This issue should be probably taken over by @jukkar or @tbursztyka. I removed myself from 'Assignees' list, won't be able to help here.

mnkp, 2018-08-07:
PR #8966 merged commits which ensure that networking tests are run in qemu_x86 or native_posix only. But we probably still need a solution for all the sample projects? If not this issue can be closed.

pfalcon, 2018-08-08:
@mnkp: Thanks for the responses! I definitely have a bunch of alternatives to present for this *could* be solved (addressed). But I'm just back from vacation and going thru the backlog, and probably straight into the 1.13 release rush, so please give me some time to prepare the response. Thanks!

pfalcon, 2018-08-10:
Ok, list of different ways this issue could be solved (@mnkp listed a few of ways in this list already - I'm glad we're on the same line). They're roughly sorted from least preferred to the most preferred.

1. Exclude boards with eth_sam_gmac from default sanitycheck runs - that's of course something which everyone wants to avoid.
2. Set the "right" number of buffers in _defconfig for a SoC/board. This option was already mentioned by @mnkp in https://github.com/zephyrproject-rtos/zephyr/issues/9015#issuecomment-406756854. The problem is that it's not flexible: ideally, we'd want to override default Kconfig setting with _defconfig, yet allow a user to override that with prj.conf, but I'm not sure it can work like that. (Ping to @ulfalizer for both question and an example of usecases users may be interested in.)
3. Move check for enough buffers from compile-time to runtime. I.e., when drivers initializes, it prints error with clear guidelines how to get it working. (This error would be disabled in "production" config build, e.g. with asserts disabled or whatever.)
4. Increase net_buf size just for eth_sam_gmac to accommodate larger size of Ethernet frame.
5. Zephyr networking stack has ability to define customer buffer pools. And that would be direct counterpart of what @mnkp writes in https://github.com/zephyrproject-rtos/zephyr/issues/9015#issuecomment-406755318 : "Drivers like eth_dw or eth_mcux reserve a private RX buffer in RAM - one continuous chunk - large enough to hold at least one full Ethernet frame." Indeed, they do, effectively they has "customer buffer pool" as a flat memory. eth_sam_gmac, being more advanced driver, can achieve the same effect by defining its own receive buffer pool, and allocate buffers for received frames strictly from it. That will ensure it can always receive a complete Ethernet frame. No special care about send buffers is required - a user who wants to send enough data in a complete frame would have enough buffers in their configs, otherwise they want to send less data at once.

As can be seen, p.5 looks like the right way to solve this issue, and directly corresponds to the "machinery" used in other drivers (just more adapted to Zephyr's way of working, up to ensuring zero-copiedness).

@mnkp, I hope you would agree with this analysis, and add such a refactor of the driver to your queue. (I for example don't have required hardware for testing. But if you think that "I develop, you test" cooperation would be beneficial for this case, we surely can give it a try.) Thanks.

jukkar, 2018-08-10:
The point 5 is not without its own problems. The gmac buffer pool would mean that the driver would only able to receive X. number of frames (X > 1) at a time. If the core IP stack is not fast enough to process the packets, we will loose packets. If global buf pool is used, then we can utilize the buffer memory better. Of course in that case we might run out of buffers too.

I do not see big issue how gmac is currently working. IMHO optimizing how gmac driver works in this respect looks a bit pointless.

pfalcon, 2018-08-29:
Ok, I didn't want to continue on this ticket during the 1.13 release rush, but #9679 and #7831 touch on it, so let's try to go thru it now again.

pfalcon, 2018-08-29:
@jukkar:

> I do not see big issue how gmac is currently working. IMHO optimizing how gmac driver works in this respect looks a bit pointless.

Here we go again, in circles. *There are* issue with sam_e70_xplained driver. It is known for at least half a year. It was subtle, so progress on analyzing and approaching it was slow, but finally this ticket was created, capturing analysis and proposing various ways to resolve it. And after all that you say that you don't see issues with it?

pfalcon, 2018-08-29:
@jukkar 

> The point 5 is not without its own problems.

I appreciate this comment too. It's akin to a philosophical statement "whatever you do, there're downsides to it". That's absolutely true. And yet we're seeking for the best compromise, and to establish the best practices for future cases.

And again, the problem we're solving is not "optimizing" sam_e70_xplained driver, but how to stop individual drivers to apply unduly settings backpressure on the entire Zephyr stack and all other boards.

pfalcon, 2018-08-29:
So, given the latest interest to this matter, @nashif kindly assigning 1.13 milestone to this ticket and @jukkar submitting https://github.com/zephyrproject-rtos/zephyr/pull/9679, I'd like to propose following plan for 1.13:

1. Once #9679 is merged, which effectively blacklists sam_e70_xplained from tests/, then,
2. I rework https://github.com/zephyrproject-rtos/zephyr/pull/7831 to have a title not starting with "Revert:", and also add sam_e70_xplained blacklisting for samples/ (there're not many cases, ~6 IIRC). Of course, the main point of that patch remains: a) getting back to reasonable buffer counts on *all* hardware; b) unbreak frdm_k64f.

Let's see if such a "compromise" can work.

The main outcome out of this all which I would expect is however: 1) we no longer change settings of NET_PKT_RX_COUNT, NET_PKT_TX_COUNT, NET_BUF_RX_COUNT, NET_BUF_TX_COUNT without very *thorough* consideration and discussion. 2) It's API contract that any Zephyr driver should work with the standard Zephyr settings of NET_PKT_RX_COUNT, NET_PKT_TX_COUNT, NET_BUF_RX_COUNT, NET_BUF_TX_COUNT. How to achieve that is up to a driver. There're whole 5 variants proposed above.

nashif, 2018-08-29:
neither #9676 not #7831 fix the issue as described in the bug, I am ok moving this issue to 1.14, was added to 1.13 in bulk, can always change this.

pfalcon, 2018-10-09:
@mnkp: Ping, is it in your queue to do something re: this ticket?

mnkp, 2018-11-06:
Thanks @pfalcon for your analysis and proposals how to solve the issue. And I apologize for a late reply: long vacations + lack of time.

I definitely agree with your statement
> And again, the problem we're solving is not "optimizing" sam_e70_xplained driver, but how to stop individual drivers to apply unduly settings backpressure on the entire Zephyr stack and all other boards.

however I don't agree with the following
> It's API contract that any Zephyr driver should work with the standard Zephyr settings of NET_PKT_RX_COUNT, NET_PKT_TX_COUNT, NET_BUF_RX_COUNT, NET_BUF_TX_COUNT.

Forcing all the drivers to use the same amount of buffers regardless of their internal architecture is very inflexible. It's akin to demanding that all I2C drivers work in fast mode setting and blacklisting all the drivers that can work in standard mode only. It's only natural that different drivers work with different configuration settings.

That said, if we are able to make drivers behave alike and have similar requirements it definitely improves portability and makes the framework easier to use. In my opinion converting sam_e70_xplained driver to use private RX buffers is a nice to have feature not a must. Especially that it makes the driver more complex.

From the discussion so far the best choice - again in my opinion - seems to be:
> 2) Set the "right" number of buffers in _defconfig for a SoC/board. [...] The problem is that it's not flexible: ideally, we'd want to override default Kconfig setting with _defconfig, yet allow a user to override that with prj.conf, but I'm not sure it can work like that.

 _defconfig is / defines the default Kconfig settings. This can be later overwritten by prj.conf files.

This approach works well if most of the applications in samples folder require 'standard' amount of RX buffers, where 'standard' is a number specific to the given SoC/board which ensures the driver works well in a typical networking scenario. Sample application which requires larger amount of buffers could simply increase the default in their prj.conf file. While possible it's unlikely that we would need to differentiate this setting for different SoCs. The only problem would be sample applications which require amount of buffers smaller than the 'standard'. But do we have any such one?

@jukkar mentioned that
> The point 5 is not without its own problems. The gmac buffer pool would mean that the driver would only able to receive X. number of frames (X > 1) at a time. If the core IP stack is not fast enough [...]

The last time I worked with Zephyr's networking stack, about a year ago, it was possible to mix and match buffer fragments from different pools in one frame. I may be wrong but I believe Ethernet driver, when looking for a free RX buffer, could get one from its own private RX buffer pool or the networking stack RX buffer pool. If this is not a case then converting the gmac driver to use private RX buffer pool would pretty much break it.

pfalcon, 2018-11-07:
@mnkp : Thanks for getting back to this.

> Forcing all the drivers to use the same amount of buffers regardless of their internal architecture is very inflexible.

But that turns the situation upside down. It's not "forcing driver to use the same amount of buffers", it's "a driver should be able to work with *any* amount of buffers, as specified by *user*". Like, that's a high ideal, to be able to *work* like that. But at least to *build* should be the baseline requirement.

> It's akin to demanding that all I2C drivers work in fast mode setting and blacklisting all the drivers that can work in standard mode only.

But again, the situation is just the opposite! ;-) The sam_e70 driver is cool, advanced, and fast - and implements only "fast mode" (== requires a lot of buffers). And now we effectively have to force everyone into "fast mode" (== a lot buffers), and cannot do CI with "standard mode" active (== not so a lot of buffers), because the sam_e70 driver breaks the build.

> From the discussion so far the best choice - again in my opinion - seems to be:

> _defconfig is / defines the default Kconfig settings. This can be later overwritten by prj.conf files.

Well, from my PoV, that's the least flexible way to deal with it. But if you think that's what can be done now, can you please proceed with it, to make sure that it all works as expected (because again, I don't have sam e70)?

jukkar, 2018-11-07:
> "a driver should be able to work with _any_ amount of buffers, as specified by _user_"

This assumption is wrong, especially with sam-e70 eth driver. There is some minimal amount of buffers that e70 needs.

I do not see why we should have a global minimum for network buffers. This clearly depends on application / board needs.

pfalcon, 2018-11-07:
> This assumption is wrong, 

That's not assumption. That's the *baseline* requirement of flexible, unmessy networking stack.

> especially with sam-e70 eth driver. There is some minimal amount of buffers that e70 needs.

Well, *networking* needs minimal amount of buffers, counted as 1 (or maybe 2). I can send data in 5-byte chunks and live long happy networking life, and should be able to do that even on sam-e70.

> I do not see why we should have a global minimum for network buffers.

We do not need to specially "have" it. It's implicit property of the world. Like, minimum number of things when they still exists is 1. Maybe for multilayer stack, we need 1 packet+buffer for each layer. So, maybe 2 or 3. That's it. 

> This clearly depends on application / board needs.

Optimum number depends on application / board. Maximum number depends on application / board. Minimum - nope. It depends only on the properties of the stack. And until the stack is able to work with absolutely minimum number of buffers, it won't be able to work reliably at all.

pfalcon, 2018-11-23:
@mnkp: Ping.

jukkar, 2019-02-19:
This was addressed in #13210

closed 2019-02-19 by jukkar