zephyrproject-rtos/zephyr #9032: net/sockets/echo_async crashes after several connections (qemu_x86)

rlubos, 2018-07-19:
I've experienced an `echo_async` sample crash when I was playing with it on qemu_x86. Actually, it's pretty simple to reproduce, just connect/disconnect a few times with telnet (for me two times is enough). Here's the log:
```
***** Booting Zephyr OS v1.12.0-880-g86b5edc4d *****
Asynchronous TCP echo server waits for connections on port 4242...
Connection #0 from 192.0.2.2 fd=4198848
poll error: 4
Connection #1 from 192.0.2.2 fd=4198848
poll error: 4
***** CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x00000000
PDE: 0x025 Present, Read-only, User, Execute Enabled
PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
Current thread ID = 0x00402180
eax: 0x00000000, ebx: 0x00401230, ecx: 0x0041521c, edx: 0x0041527c
esi: 0x00000001, edi: 0x0041527c, ebp: 0x0041520c, esp: 0x004151f8
eflags: 0x00000006 cs: 0x0008
call trace:
eip: 0x00015742
     0x00015940 (0x293)
     0x000056fd (0x415254)
     0x00003c5c (0x400340)
     0x000128ed (0x4153e8)
     0x00003e03 (0x0)
Fatal fault in essential thread! Spinning...
```

Didn't have too much time to dive in, but the crash itself takes place within `k_poll` function. Yet it looks that the problem here is that sockets are not being closed. Not sure about the error returned by the `poll` function (`EINTR` doesn't look too scary), but the fact that sockets are not closed is disturbing. Perhaps `net_context` for whatever reason is not notyfing the socket shim about connection being closed, so in result `recv` and `close` are not called?

Nothing fancy in wireshark logs, just two opened and closed connections:
[image]

rlubos, 2018-07-19:
Moreover, it looks that second `accept` returned `net_context` pointer (socket id) to the same context as the first one, even though it was still in use by the application...
So I guess it might be yet another consequence of #8008.

nashif, 2018-07-24:
@pfalcon ping

pfalcon, 2018-08-21:
Ok, I went on vacation and completely forgot about this issue. Hope to get to it soon.

pfalcon, 2018-08-21:
Reproduced with 469bd39b82cd20d732a4e5032dc8b7e895614125. Still reproduced with zephyr-v1.12.0 ;-(. Still with zephyr-v1.11.0 ;-(. Still with zephyr-v1.10.0 ;-(. Ok with zephyr-v1.9.0 (that's before cmake, old good days!).

Yeah, so that's pretty sucky a picture ;-(. Blame cmake I guess? ;-)

pfalcon, 2018-08-21:
Ok, this bisects to 48fadfe6232a4520f13da6c54a8dd204f8c5ece2. And it's dejavu and overall nonsense - of course I was in the middle of discussion leading to that change (and ones around it). What would be the explanation of that? That blocking socket handling was updated/tested after that change, but poll() implementation wasn't?

pfalcon, 2018-08-21:
"poll error: 4" is EINTR of course.

PR #9556 (2018-08-30): kernel: k_poll: Introduce separate status for cancelled events
Previously (as introduced in 48fadfe62), if k_poll() waited on a
queue (or subclass like fifo), and wait was cancelled on queue's
side using k_queue_cancel_wait(), k_poll returned -EINTR. But it
did not set event->state field (to anything else but
K_POLL_STATE_NOT_READY), so in case of waiting on multiple queues,
it was not possible to differentiate which of them was cancelled.

This in particular broke detection of network socket EOF conditions
in POSIX poll() implementation.

This situation is now resolved with introduction of ex [... cut]
commit 45c0b204 (2018-08-30): kernel: k_poll: Introduce separate status for cancelled events

Previously (as introduced in 48fadfe62), if k_poll() waited on a
queue (or subclass like fifo), and wait was cancelled on queue's
side using k_queue_cancel_wait(), k_poll returned -EINTR. But it
did not set event->state field (to anything else but
K_POLL_STATE_NOT_READY), so in case of waiting on multiple queues,
it was not possible to differentiate which of them was cancelled.

This in particular broke detection of network socket EOF conditions
in POSIX poll() implementation.

This situation is now resolved with introduction of e [... cut]
commit a2d12527 (2018-08-30): net: sockets: poll: Handle EINTR return from k_poll

This is similar to change which was done in 21f31e90ec, unfortunately
this case was missed.

Fixes: #9032
closed 2018-08-30 by nashif