zephyrproject-rtos/zephyr #9041: Some optimization suggestions about IPV6

walter-xie, 2018-07-20:
1. MLD and NDP should be extracted from the ipv6.c file.
2. The net_neighbor_table state depends on the NS packet, the dependency should be broken. With RPL, the neighbor state can be detected by DIO and DAO packet.
3. In struct net_ipv6_nbr_data, the "reachable" and "send_ns" work take up relatively large memory. There should be a better way to achieve.For example: two int variables inside the struct and a timer ouside the struct.
4. The rpl.c should be split. It looks a bit huge.

rveerama1, 2018-08-03:
I am working on issues #8721 and #8724. I will consider your inputs while optimization.

rveerama1, 2018-10-18:
IPv6 related issues are addressed and deprecating RPL (https://github.com/zephyrproject-rtos/zephyr/pull/10679). So closing this.

closed 2018-10-18 by rveerama1