zephyrproject-rtos/zephyr #9043: New logging subsystem's timestamps wrap a little before the 3-minute mark

jhedberg, 2018-07-20:
The log format implies that it can show up many hours, however it seems to be wrapping back to zero a little bit before the three minute mark.

nordic-krch, 2018-07-20:
@jhedberg logger is using k_cycle_get_32() by default. So maybe it is wrapping that often? You can try disabling timestmap formatting (LOG_BACKEND_UART_FORMAT_TIMESTAMP) then timestamp is printed in raw format. If it still wraps it's on k_cycle_get_32().

pizi-nordic, 2018-08-03:
I have verified the  k_cycle_get_32() and I did not noticed any wrapping around there minutes.

Code snipped which I used for verification:
```
        for (int i = 0; i < 60*60; i++) {
                u32_t c = k_cycle_get_32();

                printk("%04u => %02u:%02u (0x%08X)\n",
                        i,
                        (c / sys_clock_hw_cycles_per_sec) / 60,
                        (c / sys_clock_hw_cycles_per_sec) % 60,
                        c);

                k_sleep(1000);
        }
```

Result:
```
[[[ s: 1202621 m: 1202621 e: 1202621 c: 50 ]]]
[[[ s: 1202757 m: 1202757 e: 1202757 c: 52 ]]]
[[[ s: 1202892 m: 1202893 e: 1202893 c: 50 ]]]
[[[ s: 1203028 m: 1203028 e: 1203028 c: 50 ]]]
[[[ s: 1203164 m: 1203164 e: 1203164 c: 52 ]]]
[[[ s: 1203300 m: 1203300 e: 1203300 c: 50 ]]]
[[[ s: 1203436 m: 1203436 e: 1203436 c: 52 ]]]
[[[ s: 1203572 m: 1203572 e: 1203572 c: 50 ]]]
[[[ s: 1203708 m: 1203708 e: 1203708 c: 52 ]]]
[[[ s: 1203843 m: 1203843 e: 1203843 c: 50 ]]]
[[[ s: 1203979 m: 1203979 e: 1203979 c: 50 ]]]
[[[ s: 1204115 m: 1204115 e: 1204115 c: 52 ]]]
[[[ s: 1204251 m: 1204251 e: 1204251 c: 50 ]]]
[[[ s: 1204387 m: 1204387 e: 1204387 c: 52 ]]]
[[[ s: 1204523 m: 1204523 e: 1204523 c: 50 ]]]
[[[ s: 1204659 m: 1204659 e: 1204659 c: 50 ]]]
[[[ s: 1204794 m: 1204794 e: 1204794 c: 52 ]]]
[[[ s: 1204930 m: 1204930 e: 1204930 c: 50 ]]]
[[[ s: 1205066 m: 1205066 e: 1205066 c: 52 ]]]
[[[ s: 1205202 m: 1205202 e: 1205202 c: 50 ]]]
[[[ s: 1205338 m: 1205338 e: 1205338 c: 52 ]]]
[[[ s: 1205474 m: 1205474 e: 1205474 c: 50 ]]]
[[[ s: 1205609 m: 1205609 e: 1205609 c: 50 ]]]
[[[ s: 1205745 m: 1205745 e: 1205745 c: 52 ]]]
[[[ s: 1205881 m: 1205881 e: 1205881 c: 50 ]]]
[[[ s: 1206017 m: 1206017 e: 1206017 c: 52 ]]]
[[[ s: 1206153 m: 1206153 e: 1206153 c: 50 ]]]
[[[ s: 1206288 m: 1206288 e: 1206288 c: 50 ]]]
[[[ s: 1206424 m: 1206424 e: 1206424 c: 52 ]]]
[[[ s: 1206560 m: 1206560 e: 1206560 c: 50 ]]]
[[[ s: 1206696 m: 1206696 e: 1206696 c: 52 ]]]
[[[ s: 1206831 m: 1206831 e: 1206831 c: 50 ]]]
[[[ s: 1206967 m: 1206967 e: 1206967 c: 52 ]]]
[[[ s: 1207103 m: 1207103 e: 1207103 c: 50 ]]]
[[[ s: 1207239 m: 1207239 e: 1207239 c: 50 ]]]
[[[ s: 1207375 m: 1207375 e: 1207375 c: 52 ]]]
[[[ s: 1207511 m: 1207511 e: 1207511 c: 50 ]]]
[[[ s: 1207646 m: 1207646 e: 1207646 c: 52 ]]]
[[[ s: 1207782 m: 1207782 e: 1207782 c: 50 ]]]
[[[ s: 1207918 m: 1207918 e: 1207918 c: 50 ]]]
[[[ s: 1208054 m: 1208054 e: 1208054 c: 52 ]]]
[[[ s: 1208190 m: 1208190 e: 1208190 c: 50 ]]]
[[[ s: 1208325 m: 1208325 e: 1208325 c: 52 ]]]
[[[ s: 1208461 m: 1208461 e: 1208461 c: 50 ]]]
[[[ s: 1208597 m: 1208597 e: 1208597 c: 52 ]]]
[[[ s: 1208733 m: 1208733 e: 1208733 c: 50 ]]]
[[[ s: 1208869 m: 1208869 e: 1208869 c: 50 ]]]
[[[ s: 1209004 m: 1209004 e: 1209004 c: 52 ]]]
[[[ s: 1209140 m: 1209140 e: 1209140 c: 50 ]]]
[[[ s: 1209276 m: 1209276 e: 1209276 c: 52 ]]]
[[[ s: 1209412 m: 1209412 e: 1209412 c: 50 ]]]
[... 1737 lines cut ...]
Handler Signal value :20 for 1 times
Handler Signal value :20 for 2 times
Handler Signal value :20 for 3 times
Handler Signal value :20 for 4 times
Handler Signal value :20 for 5 times
Handler Signal value :20 for 6 times
Handler Signal value :20 for 7 times
Handler Signal value :20 for 8 times
Handler Signal value :20 for 9 times
Handler Signal value :20 for 10 times
Handler Signal value :20 for 11 times
Handler Signal value :20 for 12 times
0147 => 02:30 (0x004b24cc)
0148 => 02:31 (0x004ba5cf)
0149 => 02:32 (0x004c26d2)
0150 => 02:33 (0x004ca7d5)
(...)
0176 => 02:59 (0x0059c223)
0177 => 03:00 (0x005a4326)
0178 => 03:01 (0x005ac429)
(...)
0205 => 03:28 (0x00685f7a)
0206 => 03:29 (0x0068e07d)
0207 => 03:30 (0x00696180)
```

jhedberg, 2018-08-21:
Update: it appears the issue only occurs in qemu (I only tested qemu_x86). I cannot reproduce this e.g. on an nRF target.

aurel32, 2018-10-28:
I got the same issue on an Atmel SAM E70 board, it wraps every 14 seconds. The CPU runs at 300MHz, and `k_cycle_get_32()` returns a 32-bit value. And 2**32/300e6 = 14.3 seconds. So k_cycle_get_32 does not seem a good time source for timestamping.

nordic-krch, 2018-10-29:
@aurel32 by default logger uses `k_cycle_get_32()` but it can be changed runtime using `log_set_timestamp_func` (log_ctrl.h).

pizi-nordic, 2018-12-07:
Could we close this?
It looks that whole thing is caused by high frequency of hardware timer on some architectures. There is a way to mitigate this in the logger (just provide other clock reference).

jhedberg, 2018-12-07:
@pizi-nordic sure, though applications shouldn't be required to fix this, rather the commonly used boards that have this issue (e.g. qemu_x86) should come with a solution built-in and enabled by default.

aurel32, 2018-12-07:
I think the issue is present on all ARM boards except the nRF52 based ones. The standard timer on ARM board is the ARM systick one, which usually runs in the range 10-500MHz.

jukkar, 2018-12-07:
In frdm-k64f the counter wraps after 35 sec. I agree with @jhedberg here, there should be no need to application to "fix" this. In a way this is not really a logging subsystem issue but it is easy to see the problem when looking the logging output.

pizi-nordic, 2018-12-07:
So the only generic solution I see is introduction of "k_cycle_get_64()". Otherwise every soc will have to provide some custom clock source for logging only.

pabigot, 2018-12-11:
see also #6498

nordic-krch, 2018-12-13:
i'm not sure about using `k_cycle_get_64()`. Takes more precious cycles and more memory in log message (current log message is pretty with 32 bytes). What about applying compile time known, power-of-2 divisor (shift) to `k_cycle_get_32()`.

Based on `SYS_CLOCK_HW_CYCLES_PER_SEC` logger kconfig can derive hidden config like:
```
config LOG_SYS_CLOCK_DIV
   int
   default 1024 if SYS_CLOCK_HW_CYCLES_PER_SEC>16000000
  ...
```

then timestamp function would be:
`return k_cycle_get_32() / CONFIG_LOG_SYS_CLOCK_DIV`. 

Practically, no performance penalty and period is extended.

aurel32, 2018-12-13:
That won't help, given `k_cycle_get_32()` is actually the one wrapping very quickly in less than a minute on board with a fast CPU clock. If we want to keep e 32-bit value for the logs, what about using `k_uptime_get_32()` instead? It only has a millisecond granularity, but wraps only in 49 days, which is probably acceptable for the logs.

nordic-krch, 2018-12-13:
@aurel32 it's clear now that i've probably returned from sick leave too early :)

jhedberg, 2018-12-13:
> If we want to keep e 32-bit value for the logs, what about using k_uptime_get_32() instead? It only has a millisecond granularity, but wraps only in 49 days, which is probably acceptable for the logs.

Wouldn't it be possible to use a combination of those two, i.e. k_uptime_get_32() to get the milliseconds and k_cyckle_get_32() to get the microseconds?

aurel32, 2018-12-13:
@jhedberg, I am not sure there is an easy relation between `k_uptime_get_32()` and `k_cycle_get_32()`. In addition to that, if the logger is limited to a 32-bit timestamp value, there is a trade-off to do between maximum time before wrapping and accuracy. A 32-bit value with microsecond resolution wraps in a bit more than 1 hour.

jhedberg, 2018-12-13:
@aurel32 right, that's indeed an issue. I was thinking microseconds would always be wanted since the logger renders them in the output by default. Probably this all needs to be configurable, i.e. does the user want microsecond accuracy but is willing to accept a wrap-around in a bit more than an hour, or is millisecond accuracy sufficient.

carlescufi, 2019-03-05:
@nordic-krch and @pizi-nordic. Is there a possible workaround for this issue as it stands, or do we need additional functionality in the logging subsystem?

nordic-krch, 2019-03-05:
@carlescufi `k_uptime_get_32` could be used instead. It's in milliseconds so resolution would be smaller. I would prefer to keep `k_cycle_get_32` as long as `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC` gives accepted resolution (like 32k in nrf) as this will be faster.

carlescufi, 2019-03-05:
@nordic-krch Thanks for the reply. What you propose is reasonable. Since we are trying to close the open bugs that are important to users for 1.14, would you mind sending a PR that does what you describe?

nordic-krch, 2019-03-05:
@carlescufi sure, we'll do that.

PR #14074 (2019-03-06): logging: Use k_uptime_get_32 for high frequency system clock
If system clock runs fast k_cycle_get_32(), which is the timestamp
source, wraps often (few minutes). This patch changes default
timestamp function to use k_uptime_get_32() if system clock
frequency is higher than 1 MHz and k_cycle_get_32() otherwise.

If system clock runs at 1 MHz, counter will wrap every 71.5 minutes.

Fixes #9043.
Fixes #14010.
closed 2019-03-06 by carlescufi