zephyrproject-rtos/zephyr #9044: commit c12fa740a254e5515a46e5437c9e1d263dfe99f0 breaks logging

jhedberg, 2018-07-20:
I'm not getting any output after commit c12fa740a254e5515a46e5437c9e1d263dfe99f0

Reverting it makes things work again.

nordic-krch, 2018-07-20:
@jhedberg I've tried sample/subsys/logging/logger on nrf52_pca10040 and I see the output. What example are you running? Can you give more details so I could try to reproduce? Ideally, something that I can run on nrf52_pca10040

jhedberg, 2018-07-20:
@nordic-krch this is with any Bluetooth sample with my patches to convert the Bluetooth subsystem to use the new logging APIs. The code is currently in my "log" branch: https://github.com/jhedberg/zephyr/tree/log

Note that the branch has the problematic patch reverted, so you'll want to remove the revert commit first before trying to reproduce.

jhedberg, 2018-07-20:
Is there any other info I can provide to help move this forward?

The way the logging works with my changes is by modifying `subsys/bluetooth/common/log.h` to do this:

```
#if BT_DBG_ENABLED
#define LOG_LEVEL LOG_LEVEL_DBG
#else
#define LOG_LEVEL CONFIG_LOG_DEFAULT_LEVEL
#endif

#include <logging/log.h>

LOG_MODULE_REGISTER();

#define BT_DBG(fmt, ...) LOG_DBG("%s: " fmt, __func__, ##__VA_ARGS__)
#define BT_ERR(fmt, ...) LOG_ERR("%s: " fmt, __func__, ##__VA_ARGS__)
#define BT_WARN(fmt, ...) LOG_WRN("%s: " fmt, __func__, ##__VA_ARGS__)
#define BT_INFO(fmt, ...) LOG_INF(fmt, ##__VA_ARGS__)
```

The example logs which I mentioned as being missing with the shell app come from `subsys/bluetooth/host/hci_core.c` which does this:

```
#define BT_DBG_ENABLED IS_ENABLED(CONFIG_BT_DEBUG_HCI_CORE)
#define LOG_MODULE_NAME bt_hci_core
#include "common/log.h"
```

And then later e.g.:
```
        BT_INFO("Identity: %s", bt_addr_le_str(&bt_dev.id_addr));
```

Do you see anything wrong with the above?

nashif, 2018-07-20:
is this now fixed?

jhedberg, 2018-07-20:
> is this now fixed?

Yes, apparently @nordic-krch forgot to include a reference to this issue in the PR so this wasn't auto-closed. Closing it now.

closed 2018-07-20 by jhedberg