zephyrproject-rtos/zephyr #9045: A resource-saving programming model

walter-xie, 2018-07-20:
The more threads created, the more stack memory is needed. For example: samples/net/rpl-mesh-qemu/root, it takes nearly 20kB RAM. 
noinit          0x0000000020000c70     **0x4fc0**
 *(.noinit)
 *(.noinit.*)
 .noinit."/disk/zephyr/subsys/shell/shell.c".0
                0x0000000020000c70      0x7d0 libzephyr.a(shell.c.obj)
 .noinit."/disk/zephyr/subsys/net/ip/net_pkt.c".0
                0x0000000020001440      0xbe0 subsys/net/ip/libsubsys__net__ip.a(net_pkt.c.obj)
                0x0000000020001440                _k_mem_slab_buf_rx_pkts
 .noinit."/disk/zephyr/subsys/net/ip/net_pkt.c".1
                0x0000000020002020      0xbe0 subsys/net/ip/libsubsys__net__ip.a(net_pkt.c.obj)
                0x0000000020002020                _k_mem_slab_buf_tx_pkts
 .noinit."/disk/zephyr/subsys/net/ip/net_pkt.c".2
                0x0000000020002c00      0x180 subsys/net/ip/libsubsys__net__ip.a(net_pkt.c.obj)
 .noinit."/disk/zephyr/subsys/net/ip/net_pkt.c".3
                0x0000000020002d80      0x7d0 subsys/net/ip/libsubsys__net__ip.a(net_pkt.c.obj)
 .noinit."/disk/zephyr/subsys/net/ip/net_pkt.c".4
                0x0000000020003550      0x180 subsys/net/ip/libsubsys__net__ip.a(net_pkt.c.obj)
 .noinit."/disk/zephyr/subsys/net/ip/net_pkt.c".5
                0x00000000200036d0      0x7d0 subsys/net/ip/libsubsys__net__ip.a(net_pkt.c.obj)
 .noinit."/disk/zephyr/subsys/net/ip/net_tc.c".0
                0x0000000020003ea0      0x4b0 subsys/net/ip/libsubsys__net__ip.a(net_tc.c.obj)
                0x0000000020003ea0                tx_stack
 .noinit."/disk/zephyr/subsys/net/ip/net_tc.c".1
                0x0000000020004350      0x5dc subsys/net/ip/libsubsys__net__ip.a(net_tc.c.obj)
                0x0000000020004350                rx_stack
 *fill*         0x000000002000492c        0x4 
 .noinit."/disk/zephyr/subsys/net/ip/net_mgmt.c".0
                0x0000000020004930      0x200 subsys/net/ip/libsubsys__net__ip.a(net_mgmt.c.obj)
                0x0000000020004930                mgmt_stack
 .noinit."/disk/zephyr/kernel/init.c".1
                0x0000000020004b30      0x400 kernel/libkernel.a(init.c.obj)
                0x0000000020004b30                _main_stack
 .noinit."/disk/zephyr/kernel/init.c".2
                0x0000000020004f30      0x100 kernel/libkernel.a(init.c.obj)
                0x0000000020004f30                _idle_stack
 .noinit."/disk/zephyr/kernel/init.c".3
                0x0000000020005030      0x800 kernel/libkernel.a(init.c.obj)
                0x0000000020005030                _interrupt_stack
 .noinit."/disk/zephyr/kernel/system_work_q.c".0
                0x0000000020005830      0x400 kernel/libkernel.a(system_work_q.c.obj)
                0x0000000020005830                sys_work_q_stack
 *(.kernel_noinit.*)
 
 
That would be better if we can reduce the RAM to less than 15kB, by merging associated threads. 
I think the following programming model helps achieve the goal. This makes zephyr perform well on resource-constrained SOCs.

struct k_evt {
	_wait_q_t wait_q;
	unsigned int evt_id;
};

__syscall void k_evt_init(struct k_evt *pevt);

/*and_mode: when true,all bits must be set; when false, any bit must be set*/
__syscall u32_t k_evt_get(struct k_evt *pevt, u32_t evt_mask, s32_t wait_timeout, bool and_mode=false);

__syscall void k_evt_set(struct k_evt *pevt, u32_t evt);

#define EVT_A   (1)
#define EVT_B   (2)
#define EVT_C   (4)
..............
#define EVT_N   (0x10000000)

struct k_evt evt_A;
struct k_evt evt_B;

void threadA(void)
{
	u32_t evt;
	for(;;) {
		evt = k_evt_get(&evt_A, EVT_A | EVT_B | EVT_C | ... | EVT_N, K_SECONDS(3), false);
		if (!evt){
			/*wait timeout*/
			......
		}else {
			if (evt & EVT_A) {
				.........
			}
			if (evt & EVT_B) {
				.........
			}
			........
			if (evt & EVT_N) {
				.........
			}
		}
		.......
	}
}

void threadB(void)
{
 	u32_t evt;
	for(;;){
		evt = k_evt_get(&evt_B, EVT_A | EVT_B | EVT_C , K_SECONDS(3), true);
		if (!evt){
			/*wait timeout*/
			......
		}else {
			if (evt & (EVT_A | EVT_B | EVT_C)) {
				.........
				k_evt_set(&evt_A, EVT_N);
			}
		}
		......
	}
}

void isr_procA(void)
{
	..........
	k_evt_set(&evt_A, EVT_B);
}

void isr_procB(void)
{
	..........
	k_evt_set(&evt_B, EVT_A | EVT_B | EVT_C);
}

void main(void)
{
	k_evt_init(evt_A);
	k_evt_init(evt_B);
	/*Create threadA*/
	/*Create threadB*/
	..........
}

FYI:
https://github.com/LiteOS/LiteOS/blob/master/kernel/include/los_event.h
(Keil uVision4 tool) Keil\ARM\RL\RTX\SRC\ARM\rt_Event.[ch]

hongshui3000, 2018-07-23:
The programming model I use：
https://github.com/steve-scott/fsm.git

tbursztyka, 2018-07-23:
Use make ram_report and make rom_report to get a precise count of ram/rom generated by the sample on your target.

Current network stack is known to be greedy, there is on-going work to optimize rom and ram usage.
Threads are not what takes most of the memory actually.

walter-xie, 2018-07-24:
@tbursztyka Thanks. The "make ram_report and make rom_report " command is very helpful. 
However, the total cost of the stack should also be considered. Ram is very limited on some soc chips, for example, less than 20kB. By reducing the total number of threads, we can save some ram.

By the way. With the "make ram_report" command, I found some structures that should not take up ram. For example: "struct device_config" and  "struct device " in $ZEPHYR_BASE/include/device.h L104 L109 L146 L152, they should be decorated with "const".

walter-xie, 2018-07-24:
@hongshui3000 I think FSM is better suited as application logic scheduling, rather than dividing tasks or threads. And I don't think it can save ram memory.

tbursztyka, 2018-07-25:
@walter-xie the device not being const is a known issue. Some drivers play with device's api attribute (setting it or not at runtime etc...). There is an long-shot on-going work to revise the device driver model entirely (see issue #6293)

About reducing threads, as long as it's in your application sure. In the kernel it's another story. 
On net stack you can avoid net mgmt one with CONFIG_NET_MGMT_EVENT=n
(unless you need these events).

But seriously, net stack's RAM usage is way more into how buffers are managed and object size than due to the 2-3 threads there. For instance, just right now I sent a patch saving 104 bytes per-net_if when dhcpv4 is enabled. And in IPv6/RPL: There is a _lot_ to gain as well doing the same: getting rid of the k_delayed_work per-net_if on relevant objects. (for instance, rpl instance object can be reduced by 200 bytes at least)
There are issue opened about all of this already and some work already going on.

tbursztyka, 2018-07-27:
@walter-xie for k_evt, you would need to create a new issue where you describe the object and summarize the concept and ask for comments/ideas. Make it generic (not net stack or sample oriented). If you are willing to implement it, that will be even better.

About net stack and threads, protocols themselves run mostly in 2 threads: RX and TX, timeouts on some objects run in sys_work_q thread. If you add on top of that the mgmt events, then it will be in total 4 threads. We could probably reduce these, at least thg mgmt one it could be possible to make it run in the RX thread I think. There are rooms for optimization.

pdunaj, 2018-08-06:
It's good to see that other people see the thread stacks issue.

rodrigopex, 2022-11-17:
> Maybe zbus could help here, @rodrigopex ?

For sure, @gmarull. Thanks for sharing. Zbus provides precisely what you expect, @walter-xie. This model https://github.com/zephyrproject-rtos/zephyr/issues/9045#issue-343019861 is possible with Zbus. If you need help implementing that, reach out to the zbus Discord channel.

gmarull, 2022-11-17:
Let's close this issue then, if you encounter any issues with Zbus to solve your usecase, please open a new issue.

closed 2022-11-17 by gmarull