zephyrproject-rtos/zephyr #9047: Interrupt APIs.

oxalistriangularis, 2018-07-20:
I notice that zephyr provides mechanisms to declare, enable, disable IRQ. However, there is abstracted method to set and clear IRQs. To me, it appears that we fell short of creating all the necessary abstractions for using an IRQ. I see NVIC_SetPendingIRQ and NVIC_ClearPendingIRQ called directly in the modules using direct IRQ. This implies that modules are bound ARM architecture?

Or, may be I am misunderstanding the intention?

carlescufi, 2018-07-20:
>This implies that modules are bound ARM architecture?

Yes, because pending and clearing pending interrupts is Arm-specific AFAIK
@ioannisg @pizi-nordic @cvinayak can you confirm?

cvinayak, 2018-07-20:
@oxalistriangularis @carlescufi Many other CPU architecture like x86_64 implement a driver for interrupt controller and these CPU just have a single line into CPU to put them in interrupt/exception state and a end-of-interrupt instruction. Hence, a peripheral driver on a SoC will need to use the right interrupt controller interfaces of its CPU arch, in case of ARM we use NVIC interfaces.

In x86, its either of these:
https://github.com/zephyrproject-rtos/zephyr/blob/master/include/drivers/ioapic.h
https://github.com/zephyrproject-rtos/zephyr/blob/master/include/drivers/loapic.h
https://github.com/zephyrproject-rtos/zephyr/blob/master/include/drivers/mvic.h
https://github.com/zephyrproject-rtos/zephyr/blob/master/include/drivers/sysapic.h

oxalistriangularis, 2018-07-20:
Well, what I am wondering is why don't we define and use irq_set and irq_clear APIs. Use of such abstraction would make all modules portable in the true sense.

cvinayak, 2018-07-21:
I had sent a patch in the early days, only the is_enabled interface got accepted. I was made to convince myself that this irq_set and irq_clear is very ARM/NVIC specific and other architectures do not have a way for software to set or clear a hardware (logic levels) interrupt in their external programmable interrupt controllers (PIC), like Intel 8259 PIC chips. Software Interrupts were software instructions and hardware interrupts where logic levels into PIC.

nashif, 2018-07-21:
@andrewboie FYI

oxalistriangularis, 2018-07-23:
May be then the SW interrupts offered by ARM require a better conceptual abstraction?

closed 2019-07-30 by ioannisg