zephyrproject-rtos/zephyr #9054: Build failures with mimxrt1050_evk board

nashif, 2018-07-20:
```
In file included from /home/nashif/Work/github/zephyr/drivers/serial/uart_mcux_lpuart.c:11:0:
/home/nashif/Work/github/zephyr/ext/hal/nxp/mcux/drivers/imx/./fsl_lpuart.h: In function ‘LPUART_GetDataRegisterAddress’:
/home/nashif/Work/github/zephyr/ext/hal/nxp/mcux/drivers/imx/./fsl_lpuart.h:500:30: error: ‘LPUART_Type {aka struct <anonymous>}’ has no member named ‘data’
     return (uint32_t) & (base->DATA);
                              ^~
/home/nashif/Work/github/zephyr/ext/hal/nxp/mcux/drivers/imx/./fsl_lpuart.h: In function ‘LPUART_WriteByte’:
/home/nashif/Work/github/zephyr/ext/hal/nxp/mcux/drivers/imx/./fsl_lpuart.h:602:9: error: ‘LPUART_Type {aka struct <anonymous>}’ has no member named ‘data’
     base->DATA = data;
         ^~
/home/nashif/Work/github/zephyr/ext/hal/nxp/mcux/drivers/imx/./fsl_lpuart.h: In function ‘LPUART_ReadByte’:
/home/nashif/Work/github/zephyr/ext/hal/nxp/mcux/drivers/imx/./fsl_lpuart.h:624:21: error: ‘LPUART_Type {aka struct <anonymous>}’ has no member named ‘data’
         return (base->DATA & 0x7F);
                     ^~
/home/nashif/Work/github/zephyr/ext/hal/nxp/mcux/drivers/imx/./fsl_lpuart.h:628:20: error: ‘LPUART_Type {aka struct <anonymous>}’ has no member named ‘data’
         return base->DATA;
                    ^~
/home/nashif/Work/github/zephyr/ext/hal/nxp/mcux/drivers/imx/./fsl_lpuart.h:633:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
make[3]: *** [zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/build.make:63: zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_mcux_lpuart.c.obj] Error 1
```

nashif, 2018-07-23:
for example on tests/kernel/queue as seen in https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/19349/3/tests

nashif, 2018-07-24:
board disable in sanitycheck until this is fixed: #9092

MaureenHelm, 2018-08-03:
Fixed by #9253

closed 2018-08-03 by MaureenHelm