zephyrproject-rtos/zephyr #9055: Nordic PCA10059 support tracker

larsgk, 2018-07-20:
A collection of pull requests (and issues) related to PCA10059 (Dongle) suport:

- [ ] DNM Add nRF52840 USBD driver ( #8787 )
- [ ] MCUboot nrf52840_pca10059 support ( https://github.com/runtimeco/mcuboot/pull/318 )
- [x] Add add nrf52840_pca10059 board support ( #8812 )

/cc @lemrey @nvlsianpu

carlescufi, 2018-07-20:
@larsgk thanks for creating this.

carlescufi, 2018-07-20:
cc @jhedberg

lemrey, 2018-07-20:
https://github.com/pawelzadrozniak/zephyr/pull/1 to be merged to #8787 .

postlund, 2018-10-26:
I'm stuck with a few of these dongles that I want to flash with zephyr, but I don't have an external debugger (yet). If I understand it correctly, once this is done, I can flash without an additional debugger but I would still need one in the first place to place mcu_boot on the target, right? Using DFU-flash is not sufficient?

carlescufi, 2018-10-26:
@postlund there is an issue with the current USB driver that @pawelzadrozniak. I think then that @lemrey needs to update the documentation on the board in order to describe how to use Zephyr without an external debugger.

postlund, 2018-10-26:
Perfect, thank you!

carlescufi, 2018-10-30:
@postlund @larsgk can you retry with current master? a fix was just merged https://github.com/zephyrproject-rtos/zephyr/pull/10934

carlescufi, 2018-10-30:
@lemrey it looks like all the features are in place for this now. We just need to document how to transition from the built-in bootloader to MCUboot in order for people to be able to use this without a debugger.

larsgk, 2018-10-30:
I can try when I get home

tugstugi, 2018-10-31:
Is there now a documentation how to replace the stock bootloader without jlink?

carlescufi, 2018-10-31:
@tugstugi not yet, @lemrey is working on it and will report soon

Xenoamor, 2018-11-19:
[image]
I've tried uploading the latest firmware via j-link but I keep seeing this. It doesn't appear to be working on Linux either
Any idea what might be happening?

Xenoamor, 2018-11-19:
[image]
Any idea why I can't get the hci_usb sample to work on this board? It appears to fail in the same way on both windows and linux. I'm uploading to the board via j-link and the blinky example works via this method

lemrey, 2018-11-20:
Hi @Xenoamor, on this board the default peripheral used for HCI over UART is a UARTE peripheral. At the moment we are fixing an issue in the UARTE driver which is preventing the hci_uart sample to work as intented. Could you please check if it works any better with this patch https://github.com/zephyrproject-rtos/zephyr/pull/11475 ?

Xenoamor, 2018-11-20:
Thanks, I'll give it a try when I'm back in the office tomorrow! Just to clarify that this is an issue I'm seeing with hci_usb and not hci_uart though

carlescufi, 2018-11-20:
@Xenoamor so this is a completely different issue, not related to using this board without a debugger? Are you flashing using J-Link?

Xenoamor, 2018-11-20:
Sorry yes I'm using a J-Link and nrfjprog. Should I make a new issue for this?

carlescufi, 2018-11-20:
@Xenoamor yes please, this is completely unrelated.

lemrey, 2018-11-20:
> Thanks, I'll give it a try when I'm back in the office tomorrow! Just to clarify that this is an issue I'm seeing with hci_usb and not hci_uart though

Sorry @Xenoamor, I had misunderstood. What I said applies for HCI over UARTE.
Follow carlescufi's advice then and create a new issue.

carlescufi, 2019-01-07:
All the required PRs have now been merged.
Thanks!

closed 2019-01-07 by carlescufi