zephyrproject-rtos/zephyr #9067: Failed tests: posix.sema and posix_checks on em_starterkit_em7d_v22

nniranjhana, 2018-07-23:
Test in tests/posix/semaphore failing on emsk platform with **Stack Check Fail** error:

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.99 (delayed boot 1000ms) *****
Running test suite test_sem
===================================================================
starting test - test_sema
Exception vector: 0x6, cause code: 0x2, parameter 0x2
Address 0x80002800
***** Stack Check Fail! *****
Current thread ID = 0x800068a8
Fatal fault in thread 0x800068a8! Aborting.
```

Test in tests/posix/posix_checks failing with similar error after starting:

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.99 (delayed boot 1000ms) *****
Running test suite test_posix_checks
===================================================================
starting test - test_sched_policy

The given policy number 0 is supported. The minimum priority is 0 and the maximum priority is 15.
The given policy number 1 is supported. The minimum priority is 0 and the maximum priority is 15.
Exception vector: 0x6, cause code: 0x2, parameter 0x2
Address 0x80001800
***** Stack Check Fail! *****
Current thread ID = 0x800058a8
Fatal fault in thread 0x800058a8! Aborting.
```

vonhust, 2018-08-03:
@ruuddw @andrewboie @nashif 

The bug is introduced in kernel: fix errno access for user mode (commit 7f4d00695929e7115529aa533b3c4b4ea13d5cf0).  

For ARC' stack check, Non-%SP based memory references may not reference the unallocated stack region,  so the access to errno will raise stack check exception.

vonhust, 2018-08-03:
@ruuddw @andrewboie @nashif 
For
` new_thread->errno_location = (int *)K_THREAD_STACK_BUFFER(stack); `

I think it's better to use the following

` new_thread->errno_location = (int *)(new_thread->stack_info.start +
                                                              new_thread->stack_info.size); `

Because K_THREAD_STACK_BUFFER(stack) may be not the top of thread stack.
In ARC, for kernel thread,  the top of kernel thread stack  is  K_THREAD_STACK_BUFFER(stack) - MPU_STACK_GUARD_SIZE.

For arch, i think it's possible to make adjustments on start address and size,  based on the defined stack and size to meet the requirement of mpu address alignment, e.g.  33 bytes of stack  aligned to 64 bytes of stack .

So  the information in thread->stack_info  is the final, correct, adjusted information about thread stack.

andrewboie, 2018-08-11:
I can take this, I'll move the errno location to the other end of the stack.

ruuddw, 2018-08-13:
#9264 fixes it by excluding the errno from the stack checking bounds (though some review comments stil pending on that one). Moving errno to the other end probably solves it as well.

andrewboie, 2018-08-13:
I think we should move it to the other end of the stack, it's a safer solution for all arches, the current design could introduce corruption if the stack pointer gets pushed up to, but does not exceed, the top.

PR #9458 (2018-08-17): kernel: userspace: reserve stack space to store local data
This enables reserving little space on the top of stack to store
data local to thread when CONFIG_USERSPACE. The first customer
of this is errno.

Note that ARC, due to how it lays out the user stack and
privilege stack, sets the pointer itself rather than
relying on the common way.

Fixes: #9067
commit fc182430 (2018-08-17): kernel: userspace: reserve stack space to store local data

This enables reserving little space on the top of stack to store
data local to thread when CONFIG_USERSPACE. The first customer
of this is errno.

Note that ARC, due to how it lays out the user stack and
privilege stack, sets the pointer itself rather than
relying on the common way.

Fixes: #9067
closed 2018-08-17 by nashif