zephyrproject-rtos/zephyr #9069: Convert Bluetooth subsystem to use the new logger APIs

jhedberg, 2018-07-23:
The Bluetooth subsystem should be updated to use the new LOG_* APIs. Initial work has been completed, but there are a few pending challenges:

 - The log processing thread and message buffers take close to 2k of RAM in the default configuration, causing applications which are that close to the limits to stop building.
 - Many (probably most) string parameters (`%s`) come from strings which may not be valid by the time that the log processing thread consumes them.

I don't have a good solution to the first issue. For the second one, my idea would be to add a new API to the logging subsystem that would take a string as parameter, allocate a new string and copy the original to it. Once the log message gets processed, the logging subsystem would recognize that a string came from its own memory pool, and free it. The logging calls would look something like the following:
```
        LOG_DBG("%s", LOG_STR(my_string));
```

jhedberg, 2018-07-23:
Current work for the conversion is available in my log branch: https://github.com/jhedberg/zephyr/tree/log

Another thing I've been investigating is the stack footprint of LOG_* calls. As you can see in my branch, I have a patch to remove the BT_STACK() macro, which used to add 300 bytes to the stack sizes in order to cover for the footprint of printk(). I would have assumed that since the new logging subsystem does the processing in a deferred manner, and doesn't even try to parse the format string, that it would have a negligible stack impact. However, in my testing I've actually seen an up to 200 bytes difference between no logs and logs, causing some finely tuned stacks to overflow (in particular I've seen this with mesh apps and the RX thread). Any idea why that might be?

If I would be able to remove the 300 byte BT_STACK_EXTRA, it would almost cover for the memory increase caused by the logging subsystem itself (the first item I listed in this issue).

nordic-krch, 2018-07-23:
One thing that comes to my mind is this printk https://github.com/zephyrproject-rtos/zephyr/blob/d74a56bd63440bac02372bc1dea96e40c4491c1e/subsys/logging/log_core.c#L151

This function does string formatting and use buffer allocated on stack. It is used if printk is redirected to the logger (CONFIG_LOG_PRINTK)

jhedberg, 2018-07-23:
@nordic-krch thanks. There shouldn't be any direct printk() calls from the stack. Also, I just checked and CONFIG_LOG_PRINTK is disabled.

Here's an interesting observation: I just noticed that nRF51 my RX thread consumes roughly 200 bytes less than on nRF52 (the thread size is 2k, so that's a 10% difference). Is that something to be expected with Cortex M0 vs M4? Due to this, I'm only hitting the stack overflows on nRF51 (nrf51_pca10028) whereas nRF52 is fine with the BT_STACK macro removed.

jhedberg, 2018-07-23:
That said, I do get about 200 bytes increase with both nRF51 and 52 when enabling logs.

jhedberg, 2018-07-23:
Correction to the above: nRF52 consumes less than nRF51 - not the other way around

nordic-krch, 2018-07-23:
I've tried compiling sample/subsys/logging/logger with -fstack-usage flag and checked *.su files. It is more or less as expected: log_printk takes 136 bytes (128 bytes array on stack), other logger functions takes less than 32 bytes (on nrf52).

nordic-krch, 2018-07-23:
I've tried creating a task which only handles log

`void tmp_thread(void *dummy1, void *dummy2, void *dummy3)
{
	LOG_INF("hello %d %d %d %d %d %d %d %d",1,2,3,4,5,6,7,8);
	//LOG_HEXDUMP_INF(bbuf, 123);
}`

Stack analysis shows that LOG with 8 arguments takes 128 bytes of stack, hexdump 104.

jhedberg, 2018-07-23:
Question: I have calls such as `LOG_DBG("%s", bt_addr_le_str(&addr));' With new logging API the `bt_addr_le_str()` seems to always happen, regardless of log level, due to `log_printf_arg_checker()`. Is that correct? `bt_addr_le_str()` uses `snprintk()` internally, which has similar footprint as printk. This is a slight regression from our previous logging macros where the parameters would not get evaluated if the logging level was not high enough.

Seems to me like the else branch in the __LOG macro should be something like an if (0) branch, i.e. have the compiler see it but not include it in the build if the log level doesn't match.

nordic-krch, 2018-07-23:
yeah, that might be it. I will look into fixing that.

nordic-krch, 2018-07-23:
Created #9072 that should fix that.

jhedberg, 2018-07-23:
@nordic-krch thanks, I think it's a good fix, but still doesn't solve my issue (which might not be log subsystem related at all - I'm continuing investigating it). On nRF51, without logs enabled I get this:
```
recv thread stack (real size 2304):	unused 352	usage 1952 / 2304 (84 %)
```

With logs enabled I get this:
```
recv thread stack (real size 2304):	unused 200	usage 2104 / 2304 (91 %)
```

So 152 bytes difference.

nordic-krch, 2018-07-23:
152 seems big, my dummy thread when was totally empty it used 56 bytes (i assume its kernel overhead). So log should use less than 80 bytes.

jhedberg, 2018-07-23:
I got this now:
```
recv thread stack (real size 2304):	unused 136	usage 2168 / 2304 (94 %)
```

So that's a 216 byte difference. I'm strongly suspecting something within the Bluetooth subsystem at this point (that's influenced by enabling logs), so most likely this is not the logging subsystem's fault.

jhedberg, 2018-07-24:
I've now enabled -fstack-usage, and the cause is not any new function calls that grow the stack usage, rather existing functions themselves consume more stack when they have LOG_* calls in them. As an example, `bt_mesh_net_decode()` from `subsys/bluetooth/host/mesh/net.c` consumes 48 bytes with all logs disabled, but with logs enabled the consumption grows to 96. That's double the original!

jhedberg, 2018-07-24:
The only explanation that I've so far been able to come up with is the `struct log_msg_ids` that's created for every `LOG_*()` call. Since that's on the stack of the function that issues the call, it will linger on the stack even after the `LOG_*()` "call" returns. So if you have a long chain of function calls and each function tries to log several messages, this will have a pretty significant impact on stack usage. The worst I've seen so far in my testing is a 304 byte difference.

Is there something that could be done to redesign the `__LOG()` macro implementation so that it wouldn't need to create a variable on the stack of the calling function?

nordic-krch, 2018-07-24:
@jhedberg i will look into that.

jhedberg, 2018-07-24:
I've created the separate issue #9090 for string handling. As of now, I've been able to tune the configurations of our apps so that the total memory increase doesn't prevent building for the relevant boards. The substantial impact on stack consumption would be a good & important thing to fix, but it's not strictly speaking a showstopper for Bluetooth. That means that the only real showstopper is the string handling, so looking forward to find some solution for #9090

nordic-krch, 2018-07-25:
@jhedberg  could you cherry pick that https://github.com/zephyrproject-rtos/zephyr/commit/63817e8a549cf8d73530a98c1caeb80ef575c734 and try stack usage. I've removed this local variable creation so stack usage should be slightly decreased. However, the biggest stack increase comes using log with multiple arguments since on cortex_m when function has more than 4 arguments they go through stack. Also when log has more than 3 arguments, arguments are passed as array created on stack. So this change should reduce stack usage but some will remain.

jhedberg, 2018-07-25:
> could you cherry pick that 63817e8 and try stack usage

@nordic-krch how do I do that? Which tree/branch is it in? The link doesn't seem to give any hints (in fact it makes it look like it's in the upstream master branch, but that's not the case).

nordic-krch, 2018-07-25:
branch on my fork: https://github.com/nordic-krch/zephyr/tree/logger_stack_usage_reduction

jhedberg, 2018-07-25:
@nordic-krch thanks, unfortunately I don't see much difference with it. I suspect the compiler might be generating an anonymous stack variable out of the bitwise arithmetic, possibly because it fails to evaluate the statement as a build-time constant. This should be confirmed with -fstack-usage however.

jhedberg, 2018-09-12:
We also need a couple of logging backends done before we can proceed with porting the Bluetooth subsystem: native_posix and the monitor protocol (see `subsys/bluetooth/host/monitor.c`)