zephyrproject-rtos/zephyr #9090: logging: Provide a reliable way to log transient memory/strings

jhedberg, 2018-07-24:
Currently, when logging `%s` style strings, the string needs to remain valid until the log processing thread consumes the message. In most traditional cases this will not hold true. Sometimes the string is a static buffer whose contents may change before it gets processed, other times (even worse) the string resides on the call stack, resulting in complete garbage going to the log.

In the Bluetooth subsystem we have helpers such as `bt_hex()` and `bt_le_addr_str()` which use static buffers. These are used in so many places that a solution for this issue must be available before we can move to use the new logging subsystem.

My proposal would be for the logging subsystem to have a configurable memory pool that could be used for memory that needs to stay valid from the `LOG_*()` call until the log processing thread consumes it. The APIs for allocating memory could be e.g. as follows:
```
/* Allocate & duplicate a string */
char *log_strdup(const char *str);

/* Allocate memory for a log parameter */
void *log_malloc(size_t size);

/* Duplicate memory into a log parameter buffer */
void *log_memdup(const void *mem, size_t size);
```

The first one could be used directly as part of log calls, e.g:
```
        LOG_DBG("%s", log_str_dup(my_string));
```

The second one could be used inside of our existing helpers such as `bt_hex()` and `bt_le_addr_str()`, i.e. we wouldn't need to duplicate memory, rather these helpers would use `log_malloc()` to allocate the amount they need and generate the string directly into that memory location.

When the log processing thread processes a message it would check if some parameter points into its memory pool, and if so it would free this memory for reuse.

I've intentionally suggested some APIs which are not string-specific in anticipation that we could get some format string extensions similar to what Linux has, i.e. the ability to ask the formatter to directly print in hex some memory, or to print a 6 byte MAC address. This should be handled as a separate feature request & issue, but I'm mentioned it here since it's related. The Linux extensions are specified here:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/core-api/printk-formats.rst#n199

jhedberg, 2018-07-24:
One other thing: I've been assuming that these new APIs would need to be non-blocking, i.e. they would not wait for memory to become available. If memory is not currently available they could e.g. return NULL (which the processing thread needs to handle) or in the case of string duplication perhaps return a constant string literal with some intuitive content (e.g. "<log allocation failed>").

nordic-krch, 2018-07-24:
Will look into that. The major problem there is that there are multiple potential backends so it's not obvious when to free the memory without additional knowledge. 

cc: @pizi-nordic

andrewboie, 2018-07-24:
To support LOG apis, will there be system calls or is the whole thing expected to run in user mode?
We now support malloc() for minimal libc, perhaps just use that?

nordic-krch, 2018-07-25:
As mentioned previously, the biggest problem here is to free the buffer allocated for transient stuff. Ideally it has to be freed when message is freed.
Message is freed in 2 cases: 
- when last backend processed it
- when message is discarded because message pool is empty (overwriting)

One possibility is to use dedicated memory pool (memslab) for allocating (log_strdup) then when message is freed logger checks all arguments and if any matches start of any buffer in memslab then message is freed. It will do the trick for most of the times but there are drawbacks:
- fixed slab size for application
- if message argument accidently matches memslab address but isn't memslab address then system gets messed up.
- message freeing takes more time (run through arguments, free buffers) and message freeing can happen in context of log API call (any context then) when message pool is empty (CONFIG_LOG_MODE_OVERFLOW).

Because of those drawbacks i would prefer other solution.

@pizi-nordic maybe you have something in your sleeve?

jhedberg, 2018-07-25:
@nordic-krch `k_mem_pool` might be another option? That said, both `k_mem_slab` and `k_mem_pool` come with some overhead if we don't need blocking behavior. Perhaps something more lightweight could be used because of this.

nordic-krch, 2018-07-25:
Another option might be to extend log API with macro like *LOG_EXT_INF(flags, fmt, ...)*. Flags would contain indication which arguments are allocated and must be freed with the message. 
Example:

`LOG_EXT_INF(LOG_ALLOC1 | LOG_ALLOC2, "%d %s %s", 100, log_strdup(str1), log_strdup(str2));`

In log message there are 10 bits reserved for now which can be used to create mask of allocated buffers in arguments (limiting number of arguments to 10). This solution fixes some of the drawbacks of initial proposal but extending the api. Unless somebody knows some preprocessor magic to build such mask based on standard log API.

pizi-nordic, 2018-07-25:
I think, that all this discussion goes in wrong direction. We spend a lot of time to make logger **really** fast. To get current speed, we had to sacrifice "%s". Introducing proposed changes will slow down thinks and I am afraid that we sooner or later end somewhere close to printk() speed.

IMHO for Bluetooth we could use "%02X:%02X:%02X%02X:%02X:%02X" and "%04X%04X(...)"  to dump addresses and some hex data without slowing down the logger.

jhedberg, 2018-07-25:
@pizi-nordic for addresses that might be workable. It will reduce the readability of the code significantly though, but that might be mitigated with some helper macros. What might then become a problem is the number of arguments. Isn't there some limit with logger? Most messages that log an address also log other information (sometimes two addresses). An LE address would mean 7 parameters (6 bytes + address type) so we might be quickly hitting the limits of what logger can do.

jhedberg, 2018-07-25:
@pizi-nordic also it's quite inefficient - bt_addr_le_t is 7 bytes, so that would in theory fit into two 32-bit log parameters, as long as we had a dedicated format specifier to output this correctly.

nordic-krch, 2018-07-25:
number of arguments can easily be increased to 15. More can probably also be achieved.

pizi-nordic, 2018-07-25:
@jhedberg: As far as I know logger currently supports up to 9 arguments, however there is no limit in the design. Support for additional arguments could be added.

The dedicated format for BLE address is a good idea (however we will have to figure out how to pass the address as it do not fit in single word). Also, please note that there are dedicated methods for dumping memory blobs, which could be used to replace bt_hex().

nordic-krch, 2018-07-25:
Another thing that could be done is to provide a ringbuffer that you continuously allocate from and never freed. If it is tuned correctly to application needs it works well. If overwrite occurs corrupted that is printed instead. Buffer is null terminated to ensure that %s will never go outside the buffer. Solution is simple but not reliable (can we accept that in the logger).

jhedberg, 2018-07-25:
If we follow this line of thought (accept the reduced value of `%s`), the appropriate solution for `bt_hex()` would be to use `LOG_HEXDUMP()`. There is one concern I have there: it seems there's no way to attach a descriptive string (could be a constant string literal) as metadata for the hex bytes. Take `subsys/bluetooth/host/mesh/crypto.c` as an example:

https://github.com/zephyrproject-rtos/zephyr/blob/e3bf53566e2b99973eacf2002158da08de2ec436/subsys/bluetooth/host/mesh/crypto.c#L839-L842

There would need to be a way to distinguish which value is which in the resulting log if those were converted to use `LOG_HEXDUMP()`. Is that something that could be added to the API?

jhedberg, 2018-07-25:
> Another thing that could be done is to provide a ringbuffer that you continuously allocate from and never freed. If it is tuned correctly to application needs it works well. If overwrite occurs [...]

That sounds quite similar to what we have right now with bt_hex() and bt_addr_le_str(): those have multiple buffers and cycle through them:

https://github.com/zephyrproject-rtos/zephyr/blob/e3bf53566e2b99973eacf2002158da08de2ec436/subsys/bluetooth/common/log.c#L32-L36

So we can have this solution already now without any changes in the logger, by making the number of buffers high enough (e.g. I've tried to increase the address buffers from 4 to 8 in my branch). But as you say, this is not reliable and requires careful tuning from the developer. My preference would be if we could find a more reliable solution.

pizi-nordic, 2018-07-25:
I fully support @jhedberg proposal for `LOG_HEXDUMP()` extension. IMHO adding a pointer to a string to the log message should not be a big problem.

nordic-krch, 2018-07-25:
Regarding dedicated address printing. Maybe something like this:

```
#define LOG_BLE_ADDR(addr) ((u32_t*)addr)[0], ((u32_t*)addr)[1]

LOG_INF("addr %blea data:%d", LOG_BLE_ADDR(addr), data);
```

In string formatting function %blea indicates using 2 arguments.

jhedberg, 2018-07-25:
@nordic-krch yes, I was thinking along similar lines

nordic-krch, 2018-07-25:
@pizi-nordic @jhedberg regarding hexdump and a string.
Why not use something like this:

```
LOG_INF("description of hexdump");
LOG_HEXDUMP_INF(data, length);
```

There is a possibility that output will be interleaved but since logs are prefixed it should not be a problem.
Adding string pointer is not a problem but i'm afraid that once string will be added somebody will ask for parameters to that string and that is not trivial to add and will not be effective as well.

pizi-nordic, 2018-07-25:
@nordic-krch: As logs could be dropped, we could end with description with no data or data without description.

jhedberg, 2018-07-25:
Btw, the custom format specifiers is a topic that deserves its own github issue. We should also avoid conflicts with any standard specifiers or known extensions (like the Linux ones I referred to earlier). I'd have proposed prefixing everything with %Z, but I just discovered from [printf(3)](http://man7.org/linux/man-pages/man3/printf.3.html) that it's a length specifier "A nonstandard synonym for z that predates the appearance of z.  Do not use in new code.". So some other prefix would need to be chosen.

nordic-krch, 2018-07-25:
@pizi-nordic yes, but when logs are dropped you will eventually end up in a situation where logs do not make sense.

jhedberg, 2018-07-25:
@nordic-krch I think it would be better if attaching the meta string would be atomic, i.e. done as part of a single LOG_HEXDUMP() call. Adding a string pointer to the ringbuffer that associates to the hex bytes shouldn't be that difficult. Note that dropped messages is not the only issue: with pre-emptive threads you might end up with interleaved stuff if the thread gets prempted between LOG_INF() and LOG_HEXDUMP()

nordic-krch, 2018-07-25:
@jhedberg my only issue with adding string to hexdump is that this string should be without arguments. Otherwise it makes things difficult and will degrade performance/memory utilization of the logger. So I can change hexdumps to LOG_HEXDUMP_(string, data, length) but i will insist on not making it LOG_HEXDUMP(data, length, fmt, ...).

jhedberg, 2018-07-25:
@nordic-krch for our (Bluetooth) purposes I think that's perfectly fine

jhedberg, 2018-07-25:
Regarding confusion resulting from dropped messages, we've mitigated this in the [Bluetooth monitor protocol](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/btsnoop.txt#n161) by keeping track of dropped message counts and logging a special message such as "%d messages dropped" in between normal messages. Something similar might make sense for the logging subsystem, if it doesn't have it already.