zephyrproject-rtos/zephyr #9128: doc build fails if no reST reference to file

dbkinder, 2018-07-26:
The new out-of-tree doc build process copies files to a working folder to process by copying all the reST files in the doc tree (and samples and board folders), and scanning the reST files to find references to other files (e.g., images) that need to be copied too.  This misses files that are referenced in the ``config.py`` file (e.g., logo and favicon) but not from any reST file.

Workaround is to add a comment to a reST file that appears like a reference these image files to fool the ``extract_content.py`` process.

The ``extract_content.py`` was a workaround for copying /samples and /boards documentation into the /doc tree (and referenced files) but is missing content needed for the doc build when it copies /doc to the out-of-tree build area.

dbkinder, 2018-07-26:
@carlescufi I figured out a workaround, but it was a puzzlement why the images referenced in conf.py (for the logo and favicon images) in my work on switching to use the read-the-doc theme were causing the doc build to fail.  Also, found that the _templates/ folder isn't copied over to the building directory so read-the-docs theme tweaks didn't work either.

carlescufi, 2018-07-26:
@dbkinder ok I understand, I completely missed this. If you can live with this workaround for the time being I should be able to fix it soon, although I'm on parental leave for a month.
Alternatively you can simply add the missing files to be copied to the build folder here:

https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/CMakeLists.txt#L108

Simply add a line for each file you need:

`configure_file(${CMAKE_CURRENT_LIST_DIR}/<file>${RST_OUT}/doc/<file> COPYONLY)`

Also what do you mean by `_templates/` folder? That folder doesn't exist in `doc/` ?

dbkinder, 2018-07-26:
_templates folder shows up in #9145 :)    I added a workaround.rst orphan file to reference files the copy process was missing, for now. I can tweak the CMakeLists.txt file if you think that's a better solution.

carlescufi, 2018-07-26:
@dbkinder understood. As I see it there's 2 solutions to fix this issue:

1) Maintain a list of non-rst files that we need copied into the build folder. This already exists in https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/CMakeLists.txt#L108

2) Modify `extract_contents.py` so that it copies more than just `.rst` files.  It could copy for example `.rst, .png, .html, .jpg`. This could be a command-line option to the script.

3) Use `extract_contents.py` only for `samples` and `boards` and use CMake's `copy_directory` command in script mode to make an identical copy of `doc/` into the destination build folder.

What do you think is best for the future?

dbkinder, 2018-07-26:
I'd lean towards option 3.  ``extract_contents.py`` was created to copy the documentation found in boards/ and samples/ (and any referenced images or includes) and has served us well.  It would be best to just copy the doc/ folder over to the out-of-tree location (skipping a doc/_build folder if it exists :)

carlescufi, 2018-07-26:
@dbkinder sounds good, I'll try to send a PR

dbkinder, 2018-07-26:
P.S.  It looks like the current CMakeLists.txt is copying the /samples and /boards content twice.  It should only need to copy these files into the copied /doc folder.

carlescufi, 2018-07-26:
>P.S. It looks like the current CMakeLists.txt is copying the /samples and /boards content twice. It should only need to copy these files into the copied /doc folder.

No, that's intentional. The issues is that some `.rst` files refer to code in `samples/` or `boards/` using relative paths, so we need those right below `doc/` in the build folder hierarchy

dbkinder, 2018-07-26:
ah.  ok.

PR #9307 (2018-08-06): doc: Copy full contents of doc folder
Fixes #9128
commit a0dbc7aa (2018-08-06): doc: Copy all doc/ contents and not just .rst files

To avoid having to refer to non-rst files artificially in order to get
them copied into the destination folder, make a raw full copy of the
doc/ folder instead of copying only the .rst files.

Fixes #9128
closed 2018-08-06 by carlescufi