zephyrproject-rtos/zephyr #9134: Build failure with SAM_e70 platform

kumarvikash1, 2018-07-26:
Tere are many tests across zephyr modules/components failed with build failure issue on same70.

target : same70-29k:arm
Commit : 97c06a7ab35772ef3cd6de8a39688bbd0a4dcf3c 

Error description : 
Example : 
This is a subtestcase of  /tests/kernel/workq/....
.
.
.
.

build output: [ 66%] Built target arch__arm__core
build output: [ 67%] Linking C static library libzephyr.a
build output: [ 68%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/version.c.obj
build output: [ 69%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work_q.c.obj
build output: [ 69%] Built target zephyr
build output: [ 69%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/smp.c.obj
build output: [ 70%] Linking C static library libarch__arm__core__cortex_m.a
build output: [ 70%] Built target arch__arm__core__cortex_m.a
build output: [ 72%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
build output: [ 73%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mem_domain.c.obj
build output: [ 74%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/userspace_handler.c.obj
build output: [ 74%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/userspace.c.obj
build output: [ 75%] Linking C static library libkernel.a
build output: [ 75%] Built target kernel
build output: make[1]: *** [Makefile:84: all] Error
Build failed.

nashif, 2018-07-26:
this build_output is useless. what is the build failure exactly?

nashif, 2018-07-26:
> target : same70-29k:arm

You need to use the board name in Zephyr

nashif, 2018-07-26:
ok, before filing a build error bug, always check locally and verify with master, usually build errors are resolved within hours. This was fixed already.

closed 2018-07-26 by nashif