zephyrproject-rtos/zephyr #9135: Failure : "integer overflow in exp" on Altera-Max 10 platform

kumarvikash1, 2018-07-26:
Test zephyr/tests/kernel/init fails to build and throws an error " integer overflow in exp " .

Configuration
==========
<Arch> nios2
<Board> Altera Max 10

Commit : 8ba0ddf4baeb967a03ea36bb61b150bf3890adaf

Error description : 
 zephyr/tests/kernel/init/src/main.c: In function ‘verify_bootdelay’:
build output: zephyr/tests/kernel/init/src/main.c:26:43: error: integer overflow in exp
build output:         (CONFIG_BOOT_DELAY * NSEC_PER_USEC * USEC_PER_MSEC),
build output:                                            ^
build output: zephyr/tests/ztest/include/ztest_assert.h:92:11: note: in definition of m
build output:   _zassert(cond, msg ? ("(" default_msg ")") : (default_msg), \
build output:            ^~~~
build output: zephyr/tests/kernel/init/src/main.c:25:2: note: in expansion of macro ‘za
build output:   zassert_true(SYS_CLOCK_HW_CYCLES_TO_NS64(current_cycles) >=
build output:   ^~~~~~~~~~~~

andrewboie, 2018-07-26:
Why is each line of the error message truncated? Where is the full error log?

inakypg, 2018-08-20:
Afaik yes (8b3fd6963c1b382660eac0bfdc5869dedcf745f1); setting the boot delay to 3000 (three seconds), I got an slightly different -- but same error for NIos2:

```
zephyr.git/tests/kernel/init/src/main.c:25:95: error: integer overflow in expression [-Werror=overflow] zassert_true(SYS_CLOCK_HW_CYCLES_TO_NS64(current_cycles) >=
cc1: all warnings being treated as errors
```

PR #9588 (2018-08-27): tests: kernel: init: Fix integer overflow issue
Cast the msec to nsec conversion macro with u64_t to fix the
integer overflow issue.

Fixes #9135
commit f17b111e (2018-08-27): tests: kernel: init: Fix integer overflow issue

Cast the msec to nsec conversion macro with u64_t to fix the
integer overflow issue.

Fixes #9135
closed 2018-08-27 by nashif