zephyrproject-rtos/zephyr #9148: samples/net/http_server: Failed to respond back to CURL command on http Client

tarunkum, 2018-07-27:
I was trying to test http_server from sample app made it to run on frdm_k64f boards.

The steps I have followed for http_server:

cd zephyr/samples/net/http_server
mkdir build && cd build && cmake -GNinja -DBOARD=frdm_k64f ../
ninja
cp -vr outdir/frdm_k64f/zephyr.bin /run/media/MBED/

•Here I have used frdm_k64f as http_server & the no modification is done in configuration file.

I am getting the execution failure after executing the curl command on Linux machine.
 
 curl -vvv  http://192.168.116.31/
 
 Output :- 
 
 * Rebuilt URL to: http://192.168.116.31/
 *   Trying 192.168.116.31...
 * TCP_NODELAY set
 * Connected to 192.168.116.31 (192.168.116.31) port 80 (#0)
 > GET / HTTP/1.1 
 > Host: 192.168.116.31 
 > User-Agent: curl/7.55.1 
 > Accept: */* 
 > 
 < HTTP/1.1 200 OK 
 < Content-Type: text/html 
 < Transfer-Encoding: chunked 
 < 
 <html>
   <head>
     <meta charset="UTF-8">
     <title>Zephyr HTTP server sample</title>
   </head>
 
   <body>
     <div>
       <p>It works!</p>
     </div>
   </body>
 </html>
 * Illegal or missing hexadecimal sequence in chunked-encoding
 * stopped the pause stream!
 * Closing connection 0
 curl: (56) Illegal or missing hexadecimal sequence in chunked-encoding
 ERROR-IN-SHELL

nashif, 2018-07-27:
> ERROR-IN-SHELL

what does this mean?

nashif, 2018-08-14:
@jukkar ping

jukkar, 2018-08-15:
The curl error print is fixed by #9435

pfalcon, 2018-08-21:
@tarunkum, @jukkar : Should this be closed then?

nashif, 2018-08-27:
tested, works fine.

closed 2018-08-27 by nashif