zephyrproject-rtos/zephyr #9172: Out of tree BSP

nashif, 2018-07-30:
Zephyr shall support out-of-tree BSPs (Board definition packages) with BSPs including board, soc, driver definitions and code. A BSP shall be able to have its own configuration system that integrates with the overall Zephyr configuration system.

The following shall be supported with out-of-tree feature:

- Independent applications and BSP repositories
- Ability to build applications by referencing both Zephyr and BSP
- Sanitycheck can run Zephyr tests on BSP
- Sanitycheck can test out-of-tree applications.

[image]

We already have out of tree board support, but that is done right now within an application and only supports boards. We should have the same thing for SoCs and drivers and should be able to build multiple application using this out-of-tree BSP.

nashif, 2018-07-30:
Perquisite of this is 34438

aescolar, 2018-08-01:
Having also support for out of tree archs would be very welcome for us.

nashif, 2019-02-21:
> Having also support for out of tree archs would be very welcome for us.
> (even if it would be limited to build, without sanitycheck support)

you need to open a new feature item for that. This is only for BSP

closed 2019-02-21 by nashif

aescolar, 2019-02-21:
> you need to open a new feature item for that. This is only for BSP

It is already done :)