zephyrproject-rtos/zephyr #9174: STM32L4 I2C read polling hang on NACK in stm32_i2c_msg_read()

eqvinox, 2018-07-30:
When trying to do an I2C read that is getting NACK'd in the address phase, `stm32_i2c_msg_read()` will spin forever because `RXNE` will never be set. This causes the innermost loop to spin forever.

Similar to #8915, #7626 and commit 81cfdec3a9df54b400368211c70a9a1494a8c3da, but this issue is in polling mode, not interrupt.

nashif, 2018-08-29:
@erwango will you take care of this?

ydamigos, 2018-08-30:
```
diff --git a/drivers/i2c/i2c_ll_stm32_v2.c b/drivers/i2c/i2c_ll_stm32_v2.c                                        
index 43a8a9113..2a2f5ba15 100644
--- a/drivers/i2c/i2c_ll_stm32_v2.c
+++ b/drivers/i2c/i2c_ll_stm32_v2.c
@@ -535,7 +535,9 @@ int stm32_i2c_msg_read(struct device *dev, struct i2c_msg *msg,
        len = msg->len;
        while (len) {
                while (!LL_I2C_IsActiveFlag_RXNE(i2c)) {
-                       ;
+                       if (LL_I2C_IsActiveFlag_NACK(i2c)) {
+                               goto error;
+                       }
                }
 
                *buf = LL_I2C_ReceiveData8(i2c);
@@ -546,6 +548,11 @@ int stm32_i2c_msg_read(struct device *dev, struct i2c_msg *msg,
        msg_done(dev, msg->flags);
 
        return 0;
+error:
+       LL_I2C_ClearFlag_NACK(i2c);
+       SYS_LOG_DBG("%s: NACK", __func__);
+
+       return -EIO;
 }
 #endif
``` 
@eqvinox Does the above patch fixes it?

eqvinox, 2018-08-30:
@ydamigos sorry I'll probably need about a week or so until I can test this, I'm out sick :face_with_thermometer:

psidhu, 2018-12-07:
@ydamigos I got into this state once. Your patch solved that issue.

ydamigos, 2018-12-07:
@psidhu Thanks for the feedback. I had forgot it completely.

ydamigos, 2018-12-17:
PR #12117 was merged.

closed 2018-12-17 by ydamigos