zephyrproject-rtos/zephyr #9194: generated syscall header files don't have ifndef protection

galak, 2018-07-31:
For example:

'include/generated/syscalls/rtc.h'

Doesn't have anything like:

```
#ifndef SYSCALL_RTC_H
#define SYSCALL_RTC_H

....

#endif /* SYSCALL_RTC_H */
```

andrewboie, 2018-07-31:
I'm not seeing a need to change this, what problem would this solve?

These generated headers should never be included directly by anyone, they really need to only be included at the very end of the corresponding headers that define the system calls, inside that header's ifdef guard.

You'll never be able to make these generated headers safe to include in a general sense as they also do not include the necessary headers to pull in data structure definitions used by the system calls.

galak, 2018-08-02:
I think doing this might avoid some weird error that might be hard to track down later.  The system calls have introduced some interesting include path issues (see commit 8777ff1304fd7c07f35a7aca655b92d0e98a277b).  So thinking having the headers have proper ifndef protection would just be good form.

nashif, 2020-02-13:
we do not have these headers anym,ore

closed 2020-02-13 by nashif