zephyrproject-rtos/zephyr #9198: Out-of-Tree YAML and DTS support

jli157, 2018-07-31:
Is it possible to have yaml definitions and dts files in an out-of-tree board directory? 

```
boards/arm/myboard/
                                   bindings/my_peripheral_dev.yaml
                                   my_peripheral_dev.dts
```

Thank you!

hakonfam, 2018-08-01:
Yes, use the `BOARD_ROOT` variable as specified in [here](http://docs.zephyrproject.org/application/application.html#custom-board-definition).

jli157, 2018-08-01:
BOARD_ROOT doesn't support defining out-of-tree yaml files since the script `extract_dts_includes.py` only reads yaml files from the directory "${ZEPHYR_BASE}/dts/bindings" which is specified on the line 105 of the file `${ZEPHYR_BASE}/cmake/dts.cmake`.

pizi-nordic, 2018-08-01:
FYI: @pdunaj

pdunaj, 2018-08-06:
The same issue: https://github.com/zephyrproject-rtos/zephyr/issues/8849

jli157, 2018-08-06:
Closed since the issue #8849 already covers the concerns this issue raises.

closed 2018-08-06 by jli157