zephyrproject-rtos/zephyr #9224: sam_e70_xplained fails to build several tests

galak, 2018-08-01:
Several networking tests don't build anymore due to:
```
/home/galak/git/zephyr/drivers/ethernet/eth_sam_gmac.c:79:2: error: #error Not enough RX buffers to allocate descriptors for each HW queue
 #error Not enough RX buffers to allocate descriptors for each HW queue
  ^~~~~
make[3]: *** [zephyr/CMakeFiles/zephyr.dir/build.make:297: zephyr/CMakeFiles/zephyr.dir/drivers/ethernet/eth_sam_gmac.c.obj] Error 1

```

sam_e70_xplained          tests/net/6lo/net.6lo                              FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/6lo/net.6lo/build.log

total complete:  149/ 221  67%  failed:    1

sam_e70_xplained          tests/net/app/net.app.no-ipv6                      FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/app/net.app.no-ipv6/build.log

total complete:  159/ 221  71%  failed:    2

sam_e70_xplained          tests/net/app/net.app.no-ipv4                      FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/app/net.app.no-ipv4/build.log

total complete:  160/ 221  72%  failed:    3

sam_e70_xplained          tests/net/dhcpv4/net.dhcp                          FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/dhcpv4/net.dhcp/build.log

total complete:  162/ 221  73%  failed:    4

sam_e70_xplained          tests/net/app/net.app                              FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/app/net.app/build.log

total complete:  163/ 221  73%  failed:    5

sam_e70_xplained          tests/net/context/net.context                      FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/context/net.context/build.log

total complete:  166/ 221  75%  failed:    6

sam_e70_xplained          tests/net/iface/net.iface                          FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/iface/net.iface/build.log

total complete:  167/ 221  75%  failed:    7

sam_e70_xplained          tests/net/app/net.app.with-dns                     FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/app/net.app.with-dns/build.log

total complete:  169/ 221  76%  failed:    8

sam_e70_xplained          tests/net/ip-addr/net.ip-addr                      FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/ip-addr/net.ip-addr/build.log

total complete:  174/ 221  78%  failed:    9

sam_e70_xplained          tests/net/rpl/net.rpl                              FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/rpl/net.rpl/build.log

total complete:  175/ 221  79%  failed:   10

sam_e70_xplained          tests/net/promiscuous/net.promisc                  FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/promiscuous/net.promisc/build.log

total complete:  176/ 221  79%  failed:   11

sam_e70_xplained          tests/net/route/net.route                          FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/route/net.route/build.log

total complete:  177/ 221  80%  failed:   12

sam_e70_xplained          tests/net/ptp/clock/net.ptp.clock                  FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/ptp/clock/net.ptp.clock/build.log

total complete:  178/ 221  80%  failed:   13

sam_e70_xplained          tests/net/tx_timestamp/test                        FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/tx_timestamp/test/build.log

total complete:  179/ 221  80%  failed:   14

sam_e70_xplained          tests/net/neighbor/net.neighbour                   FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/neighbor/net.neighbour/build.log

total complete:  180/ 221  81%  failed:   15

sam_e70_xplained          tests/net/mgmt/net.management                      FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/mgmt/net.management/build.log

total complete:  181/ 221  81%  failed:   16

sam_e70_xplained          tests/net/udp/net.udp                              FAILED: build_error
	see: sanity-out/sam_e70_xplained/tests/net/udp/net.udp/build.log

galak, 2018-08-01:
@tgorochowik Can you look at this.

tgorochowik, 2018-08-02:
@galak it should be ok now.