zephyrproject-rtos/zephyr #9238: Get POSIX board compliant with default configuration guidelines

erwango, 2018-08-02:
Created from #7151, dedicated to 96Boards boards

Following introduction of "default board configuration guidelines" (#6858), one part of the work is to move boards already ported to zephyr to conform with guidelines available here:
http://docs.zephyrproject.org/porting/board_porting.html#default-board-configuration

https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/arm/nucleo_f429zi could be used as reference for this update.

aescolar, 2018-08-02:
@erwango Is it not following the guideline already?

erwango, 2018-08-03:
@aescolar , I created tickets for all the boards, didn't look into each one in detail.
I had a look, and it seems it is. Since there is no pinmux, there is indeed not much to do.
One point not clear to me: I don't see any .dts file. Does it mean there won't ever be one?

aescolar, 2018-08-03:
> I don't see any .dts file. Does it mean there won't ever be one?

I guess there hasn't been a need for it. If there would be, I guess it would be added.
Note that the native_posix board is a testing target. Not a real embedded board.

erwango, 2018-08-03:
> Note that the native_posix board is a testing target. Not a real embedded board.

Yeah, sure. 
But if/once we rely on dts to get information on boards as for CI case for instance (instead of current yaml) it might be required. This is not yet the case anyway.

So let's close this ticket. Thanks!

closed 2018-08-03 by erwango