zephyrproject-rtos/zephyr #9240: Get ARM boards compliant with default configuration guidelines

erwango, 2018-08-02:
Created from #7151, dedicated to ARM (mps2, beetle, ..) boards

Following introduction of "default board configuration guidelines" (#6858), one part of the work is to move boards already ported to zephyr to conform with guidelines available here:
http://docs.zephyrproject.org/porting/board_porting.html#default-board-configuration

https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/arm/nucleo_f429zi could be used as reference for this update.

ioannisg, 2019-05-28:
@erwango the posted link does not seem to work. Are we sure this issue is still applicable?

erwango, 2019-05-28:
@ioannisg , doc chapter has been moved, here is the new link:
https://docs.zephyrproject.org/latest/guides/porting/board_porting.html#default-board-configuration

This is still applicable.