zephyrproject-rtos/zephyr #9242: Get NORDIC SoC based boards compliant with default configuration guidelines

erwango, 2018-08-02:
Created from #7151, dedicated to NORDIC boards

Following introduction of "default board configuration guidelines" (#6858), one part of the work is to move boards already ported to zephyr to conform with guidelines available here:
http://docs.zephyrproject.org/porting/board_porting.html#default-board-configuration

https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/arm/nucleo_f429zi could be used as reference for this update.

lemrey, 2018-09-12:
Are there any outstanding tasks related to this after #9551 has been merged? What about other third-party boards with Nordic SOCs that are yet to be aligned?

carlescufi, 2018-09-12:
@lemrey I don't know. Someone needs to go over the boards with the guidelines in hand to check. @anangl, @ioannisg, do you have time to look at this? This should apply to all boards that have Nordic chips in them.

carlescufi, 2018-12-21:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/12209

closed 2018-12-21 by carlescufi