zephyrproject-rtos/zephyr #9262: tests/kernel/mem_protect/userspace.access_other_memdomain fails on sam_e70_xplained

spoorthik, 2018-08-03:
**Testcase:** tests/kernel/mem_protect/userspace.access_other_memdomain

**Latest commit on which failure found:** 4b22ba7e4ba82e9f63e10eef89722047858033fc

**Execution log:**
```
***** Booting Zephyr OS v1.12.0-1187-g4b22ba7 (delayed boot 1000ms) *****                                                                                                                                          
Running test suite userspace
===================================================================
starting test - is_usermode
PASS - is_usermode
===================================================================
starting test - write_control
PASS - write_control
===================================================================
starting test - disable_mmu_mpu
***** BUS FAULT *****
  Imprecise data bus error
***** Hardware exception *****
Current thread ID = 0x204032a0
Faulting instruction address = 0x4007e2
Caught system error -- reason 0
PASS - disable_mmu_mpu
===================================================================
starting test - read_kernram
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x2040041c
***** Hardware exception *****
Current thread ID = 0x204032a0
Faulting instruction address = 0x400a26
Caught system error -- reason 0
PASS - read_kernram
===================================================================
starting test - write_kernram
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x2040041c
***** Hardware exception *****
Current thread ID = 0x204032a0
Faulting instruction address = 0x400a80
Caught system error -- reason 0
PASS - write_kernram
===================================================================
starting test - write_kernro
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x2040041c
***** Hardware exception *****
Current thread ID = 0x204032a0
Faulting instruction address = 0x400af8
Caught system error -- reason 0
PASS - write_kernro
===================================================================
starting test - write_kerntext
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x2040041c
***** Hardware exception *****
Caught system error -- reason 0
PASS - write_kerntext
===================================================================
starting test - read_kernel_data
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x2040041c
***** Hardware exception *****
Current thread ID = 0x204032a0
Faulting instruction address = 0x400b52
Caught system error -- reason 0
[... 221 lines cut ...]
starting test - read_kobject_user_pipe
syscall _handler_k_pipe_put failed check: Memory region 0x204033e4 (size 1) read access denied
***** Hardware exception *****
Current thread ID = 0x204032a0
Faulting instruction address = 0x20400201
Caught system error -- reason 0
PASS - read_kobject_user_pipe
===================================================================
starting test - access_other_memdomain

    Assertion failed at /home/kspoorth/work/latest_zephyr/tests/kernel/mem_protect/userspace/src/main.c:539: shared_mem_thread: (Reached unreachable code)
Thread accessed global in other memory domain

FAIL - access_other_memdomain
===================================================================
===================================================================
PROJECT EXECUTION FAILED

```
**Steps to reproduce:** 
source zephyr.git/zephyr_env.sh
cd zephyr.git/tests/kernel/mem_protect/userspace
mkdir build && cd build
cmake -DBORD=sam_e70_xplained .. 
make flash

ioannisg, 2018-08-14:
The test is unstable for nrf52_pca10040, as well.

AdithyaBaglody, 2018-08-14:
I have a PR #9288 up already.

ioannisg, 2018-08-14:
Yes, I have, now, seen that, @AdithyaBaglody , thanks!