zephyrproject-rtos/zephyr #9273: k_pipe_alloc_init() api is failing on qemu_x86

ajaykish, 2018-08-03:
k_pipe_alloc_init() api is failing on qemu_x86. This api is getting validated with the test case in https://github.com/zephyrproject-rtos/zephyr/pull/9231. 

Same test is passing in all supported architecture, including x86 based boards ( quark_se_c1000, arduino_101 and quark_d2000_crb).

andrewboie, 2018-08-13:
> Same test is passing in all supported architecture, including x86 based boards ( quark_se_c1000, arduino_101 and quark_d2000_crb)

None of those boards enable the MMU.

dcpleung, 2018-08-15:
There are two issues with the test code.

- First is the test_pipe_alloc() is running out of mem pool space when compiled for qemu_x86.
- Second is at test_pipe_block_writer_wait(). Calling k_pipe_block_put() when the pipe has no enough space to take in all data will cause the subsequent k_pipe_get() to cause CPU fault inside _abort_timeout(). However, when simply using k_pipe_put(), there would not be any issues with get() afterwards even with full or partially full pipe when put() is called. So this needs further investigation.

dcpleung, 2018-08-31:
Not an issue with API. The new tests being introduced are using more memory, but the PR never increases the mem_pool, and thus running out of memory.

closed 2018-08-31 by dcpleung