zephyrproject-rtos/zephyr #9310: nRF52_PCA10040: Failing test_slice_reset

cvinayak, 2018-08-06:
[2018-08-16 11:29:11.786] starting test - test_slice_reset
[2018-08-16 11:29:12.113] 
[2018-08-16 11:29:12.113]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:39: thread_tslice: t <= expected_slice[thread_idx] is false
[2018-08-16 11:29:12.125] 
[2018-08-16 11:29:12.344] FAIL - test_slice_reset
[2018-08-16 11:29:12.345] ===================================================================
[2018-08-16 11:29:12.357] starting test - test_slice_scheduling
[2018-08-16 11:29:12.560] A
[2018-08-16 11:29:12.560]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.576] 
[2018-08-16 11:29:12.581] B
[2018-08-16 11:29:12.581]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.604] 
[2018-08-16 11:29:12.604] C
[2018-08-16 11:29:12.604]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.612] 
[2018-08-16 11:29:12.612] D
[2018-08-16 11:29:12.612]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.628] 
[2018-08-16 11:29:12.628] E
[2018-08-16 11:29:12.629]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.646] 
[2018-08-16 11:29:12.646] F
[2018-08-16 11:29:12.646]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.663] 
[2018-08-16 11:29:12.663] G
[2018-08-16 11:29:12.664]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.680] 
[2018-08-16 11:29:12.681] H
[2018-08-16 11:29:12.681]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.698] 
[2018-08-16 11:29:12.698] I
[2018-08-16 11:29:12.698]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.715] 
[2018-08-16 11:29:12.716] 
[2018-08-16 11:29:12.716] 
[2018-08-16 11:29:12.716]     Assertion failed at /home/parker/workspace/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:48: thread_tslice: ((tdelta <= SLICE_SIZE) && ((int)p1 == thread_idx)) is false
[2018-08-16 11:29:12.733] 
[2018-08-16 11:29:12.733] FAIL - test_slice_scheduling
[2018-08-16 11:29:12.736] ===================================================================
[2018-08-16 11:29:12.742] starting test - test_priority_scheduling
[2018-08-16 11:29:12.746] PASS - test_priority_scheduling
[2018-08-16 11:29:12.748] ===================================================================
[2018-08-16 11:29:12.754] ===================================================================
[2018-08-16 11:29:12.761] PROJECT EXECUTION FAILED

carlescufi, 2018-08-06:
Assigning to @pizi-nordic since he's already looked into failing kernel tests

pizi-nordic, 2018-08-07:
Should be already fixed by #9073. I will confirm that as soon as I finish #9275.

pizi-nordic, 2018-08-07:
I confirmed that #9073 solves this problem.

pswarnak, 2018-08-08:
This is already reported in #8896.

carlescufi, 2018-08-09:
closing as duplicate

closed 2018-08-09 by carlescufi